steve-community/steve #448: Exception error message in log

rdc-Green, 2020-11-19:
I'm getting the following error in my log:

[ERROR] 2020-11-19 13:39:59,572 de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer - Exception occurred
com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `ocpp.cs._2015._10.ReadingContext` from String "CurrentMax": not one of the values accepted for Enum class: [Interruption.Begin, Interruption.End, Trigger, Transaction.Begin, Sample.Clock, Transaction.End, Sample.Periodic, Other]
 at [Source: UNKNOWN; line: -1, column: -1] (through reference chain: ocpp.cs._2015._10.MeterValuesRequest["meterValue"]->java.util.ArrayList[0]->ocpp.cs._2015._10.MeterValue["sampledValue"]->java.util.ArrayList[1]->ocpp.cs._2015._10.SampledValue["context"])
	at com.fasterxml.jackson.databind.exc.InvalidFormatException.from(InvalidFormatException.java:67) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.DeserializationContext.weirdStringException(DeserializationContext.java:1702) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.DeserializationContext.handleWeirdStringValue(DeserializationContext.java:947) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.EnumDeserializer._deserializeAltString(EnumDeserializer.java:257) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.EnumDeserializer.deserialize(EnumDeserializer.java:181) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:138) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:293) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:156) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:290) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:249) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:26) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:138) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:293) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:156) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:290) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:249) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:26) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:138) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:293) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:156) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.ObjectMapper._readValue(ObjectMapper.java:4497) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2730) ~[jackson-databind-2.11.3.jar:2.11.3]
	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:3183) ~[jackson-databind-2.11.3.jar:2.11.3]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.handleCall(Deserializer.java:128) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.accept(Deserializer.java:78) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:49) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:107) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:79) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketText(JettyWebSocketHandlerAdapter.java:83) ~[spring-websocket-5.2.9.RELEASE.jar:5.2.9.RELEASE]
	at jdk.internal.reflect.GeneratedMethodAccessor83.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:241) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:65) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:233) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:150) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:319) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:201) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:226) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.parseSingleFrame(Parser.java:260) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:583) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:441) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:428) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
[... cut ...]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at java.lang.Thread.run(Thread.java:834) [?:?]
[INFO ] 2020-11-19 13:39:59,572 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=NGV_0108, sessionId=ae879d77-1429-3a02-3285-08fd843c37ee] Sending: [4,"6934","FormationViolation","The payload for action could not be deserialized",{"errorMsg":"Cannot deserialize value of type `ocpp.cs._2015._10.ReadingContext` from String \"CurrentMax\": not on..."}]

Have you seen this before?

lategoodbye, 2020-11-21:
Based on the exception, i would assume that the MeterValues.Req from the OCPP client is formatted wrongly.

goekay, 2020-11-21:
> Cannot deserialize value of type ocpp.cs._2015._10.ReadingContext from String "CurrentMax": not one of the values accepted for Enum class: [Interruption.Begin, Interruption.End, Trigger, Transaction.Begin, Sample.Clock, Transaction.End, Sample.Periodic, Other]

the exception is clear. the station uses the value `CurrentMax` for `ReadingContext`. and, this is not one of the allowed values according to specification.

rdc-Green, 2020-11-21:
Thank you both for your response. I'll try and reconfigure the charge point.

closed 2020-11-21 by rdc-Green