steve-community/steve #451: Add missing / in nginx sample

manitu-opensource, 2020-11-25:
Currently there is a
`proxy_pass http://192.168.1.100:8080/steve/websocket/CentralSystemService;`
as sample for the `/steve/websocket/CentralSystemService` location.

This will fail if the OCPP client (e.g. KEBA P30) will NOT auto-add a slash (/) at the end before inserting the occp station id.

We recomment to use

`proxy_pass http://192.168.1.100:8080/steve/websocket/CentralSystemService/;`

This should fit all ocpp clients as nginx is intelligent enough to "auto-fix" double slashes if some client tries to.

csamsel, 2020-11-26:
I don't think rerwriting the URL during the proxy'ing is a good idea. The charging station should then be configured with a base url including a / at the end.

goekay, 2021-03-22:
agree with @csamsel. his suggestion looks good to me. it feels wrong to modify steve for some station's failure.

closed 2021-03-22 by goekay