steve-community/steve #463: WS-Addressing - failed to retrieve Message

GRangaRao, 2020-12-13:
[WARN ] 2020-12-12 04:28:45,551 org.apache.cxf.ws.addressing.ContextUtils - WS-Addressing - failed to retrieve Message Addressing Properties from context
[WARN ] 2020-12-12 04:28:45,551 org.apache.cxf.ws.addressing.ContextUtils - WS-Addressing - failed to retrieve Message Addressing Properties from context
[WARN ] 2020-12-12 04:28:45,551 org.apache.cxf.ws.addressing.ContextUtils - WS-Addressing - failed to retrieve Message Addressing Properties from context

Why this Warning is originating?

Both issues are similar?
https://github.com/RWTH-i5-IDSG/steve/issues/58

What is missing to overcome this error

goekay, 2020-12-13:
> Why this Warning is originating?

the station is not setting some headers that are required. ws-addressing is part of the soap spec and defines some additional headers for the message.

this is crucial for stations communicating via soap, because one of the fields (from) contains the server address of the station. it is used to execute operations from steve at this station. if it is missing, we don't know the destination of the message and therefore this station will not be included in the charging stations list.

> What is missing to overcome this error

the station should start setting these headers in the message. there is nothing steve can do.

GRangaRao, 2020-12-14:
Okay goekay, Thanks
Is it possible to eliminate this warning from the log by making any settings in the steve?

Even if this warning comes up, i see heartbeats are generated correctly. Will the charge sessions not get impacted with this warning?

csamsel, 2020-12-14:
It's a fatal error, it doesn't make sense to supress it.

goekay, 2021-02-06:
> Even if this warning comes up, i see heartbeats are generated correctly. Will the charge sessions not get impacted with this warning?

station will be able to send messages to steve, but you cannot trigger operations at the station from steve. the communication will be unidirectional only.

if you want, you can set the log level of this class `org.apache.cxf.ws.addressing.ContextUtils` to ERROR such that WARN logs are suppressed.

i consider this issue resolved and am closing it.

closed 2021-02-06 by goekay