steve-community/steve #464: How is authorization controlled through Steve

rishabhmehandru, 2020-12-17:
Hi

I do not see what key in database is used to authorize a charge transaction. Any help in this regards will be very helpful.

I am trying to link steve with a user database to authorize a charge transaction based on balance in user account. Using a credit card based transaction is one way of doing it, but I do want to have flexibility of a separate account based authorization.

goekay, 2020-12-17:
in the world of ocpp, every transaction is linked to and authorized by an `id tag` (it used interchangeably with `ocpp tag`). this value, generally, identifies an RFID card of the card holder (i.e. user/customer). the role `user` in steve does not exist in ocpp. it links the card to a user (or also multiple users) in the database.

rishabhmehandru, 2020-12-22:
Thanks! That makes sense. I can use id_tag to identify and intiate charge. However, how do I track energy used by a certain id_tag? I tried looking through transactions, but it mostly tracks time stamps for start and stop of a charging event. Where is the database field which reflects how much kWH or units were used by any particular id_tag?

goekay, 2020-12-23:
ocpp has StartTransaction and StopTransaction messages with meterStart and meterStop values. the difference between them is the Wh consumption.

i suggest you to take a look at the ocpp specification. you could have found the answers to your questions so far there :)

closed 2020-12-23 by goekay

rishabhmehandru, 2020-12-23:
THanks! again...so are start_value and stop_value in transaction_start and transaction_stop the meter values?