steve-community/steve #467: DB backup alternative - please include meter value on timestamp e-mail

gernotontheroad, 2020-12-17:
Hi everyone

**Background Story:**
I had been running Steve 3.4.3 on docker on Mac for 3 weeks and everything was running great until yesterday, when an update brought my Mac down and I had to restore it from a Time Machine backup. I had included the docker VM in time machine backup but when I restored the machine with the data from the night before the crash, the docker and Steve setup was there and running but without a single transaction data. I was also not able to locate the DB on the indicated location (/user/library/container/...) to have it recovered individually.

**What I would like to achieve:**
In order to avoid a 2nd data loss I would now like to export all transactions automatically by e-mail (and parse it into an Excel spreadsheet). I was hoping that the Steve e-mail notifications (Notify when.... a charging station starts/ends a transaction) would do the trick but they only contain the time stamp and no meter value related information.

Has anyone an idea how I could add the start and stop value of a transaction to the notification?

Thank you
Gernot

goekay, 2020-12-17:
> ... export all transactions automatically by e-mail ...

i understand from this that you do not want to use the functionality on transactions page to export and download the data as csv manually?

gernotontheroad, 2020-12-17:
Thank you for looking into it

Exactly, as I currently have no other means to backup that data I would want to export it by email after each transaction.

Regards
Gernot

goekay, 2020-12-17:
got it. i can easily modify the mail bodies of current notifications (start and stop) to include these values.

you should keep in mind: it would be best to have an automated backup strategy of the database in regular intervals (e.g. daily) instead of relying these mails/events. a lost database contains other valuable information besides transactions.

gernotontheroad, 2020-12-17:
Thank you, I will also look into the automated database dump

commit be1c8d2b (2020-12-17): add start/end transaction details to notification mails (Closes #467)
commit 91eca19b (2020-12-17): add start/end transaction details to notification mails (Closes #467)
closed 2020-12-17 by goekay