steve-community/steve #475: Plug & Charge ISO 15118

N1pp3, 2021-01-12:
Hi!

Thanks for this amazing software! 

We stared to play around with Plug & Charge and made it work whit Steve. 
But there is a smal problem. 

We had to plug in the car and then add the vehicle ID from "Unknown Tags".
Because of the limitations to add Colon (punctuation) in add new ID tag. 

example: 
VID:00XXXXXXXXXX

So if you are able to remove the punctuation limitation it will be smother to add new ID. 

We have tried it on Volvo and Audi and the issue is the same. We need to add VID: before ID, for it to recognize the vehicle.

The charger is hypercharger HYC 150. 

Kind regards
Jonas

lategoodbye, 2021-02-08:
For me it looks you describe Autocharge and not Plug & Charge. Autocharge uses the CCS MAC address of the EV as the ID.

goekay, 2021-03-22:
hey @N1pp3 

sorry, your request flew under my radar, even though it was easy to implement. i added the logic with the commit above. feel free to test it and give feedback. i will merge it into master afterwards.

goekay, 2021-03-24:
> So my question is where does this limitation on SteVe come from?

steve itself. i guess in the past we had some charging stations having troubles with more loose definitions of IdToken. therefore, we must have decided to limit the what a human user can enter as idtag. currently, the characters are restricted to: Upper or lower case letters, numbers and dot, colon, dash, underscore symbols.

lategoodbye, 2021-03-24:
Thanks for the explanation

closed 2021-03-30 by goekay