steve-community/steve #477: [ERROR] de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - Transport error

gabriel1lima, 2021-01-12:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :  steve-3.4.3
Operating system  :  Ubuntu 18.04.4 LTS
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior
The charging station remains connected to Steve but on random occasions the error seen in the attached image happens, which causes the charging station to disconnect and reconnect soon afterwards, however sometimes the connection with Steve is lost forever, having to restart the charging station.

[image]

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

csamsel, 2021-01-27:
That sounds like a connection drop. You could try to configure keep alive but if the underlying connection drops this wont help.

goekay, 2021-03-22:
> ... which causes the charging station to disconnect and reconnect soon afterwards

this exception probably is not the reason why the station disconnects, but rather the consequence.

> ... sometimes the connection with Steve is lost forever, having to restart the charging station.

steve acts as a websocket server and does not own the connection. the station owns the connection, since it is the initiator. there is nothing we can do about this.

closed 2021-03-22 by goekay