steve-community/steve #478: Eichrechtskonforme (Certified Meter) Upgrade

f8bar, 2021-01-13:
#### Checklist

- [x ] I checked other issues already and found no answer
- [x ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

Does somebody already work on an implementation to send OCMF data, PKI data from the charger to the backend in form of a modified Start/StopTransaction and MeterValue Requests?

#### Additional context
Hier die Implementierung von Allego: https://cdn.allego.eu/-/media/project/allegofrontoffice/allegoeu/legal/eve-double-pro-and-eve-double-pg.pdf?la=de-de&hash=83179F57DFF6E8F2596857D823A26E59
Hier findet man die Tranzparenzsoftware für die Validierung: https://www.safe-ev.de/de/

goekay, 2026-10-28:
hey @f8bar (if you are still around), i am starting the work on it :) should take couple of days.

f8bar, 2026-10-28:
In the meantime, I’ve switched industries. 😆 But this is still needed, so I really appreciate your efforts!

EmbryoSpace, 2026-10-29:
In case it helps with testing this: https://github.com/BSVKey/ocmf-kit generates real signed OCMF 1.4 records and the OCPP 1.6 `SignedData` sampled values that carry them, plus a fixture set (`fixtures/`, one folder per case with the records, the advertised public key and the expected verdict). Cases include valid sessions and the usual rejections: tampered reading, wrong key, invalid signature, replayed or missing page, meter swap, register going backwards, error flags. Apache-2.0, no dependencies, so the fixtures can be copied into a Java test suite as plain files.

closed 2026-11-10 by goekay