steve-community/steve #484: OCPI integration

Safari94, 2021-01-23:
Hi,

Based on interoperability between multiple OCPP servers, are you expecting to implement the OCPI web services to allow communication between ocpp servers?

If i can help somehow in the implementation please let me know.

Thanks

Kind regards.

goekay, 2021-01-26:
hello, there are no plans for implementing an OCPI integration. however, in the past we implemented similar roaming integrations, one for Hubject and [one for Plugsurfing](https://github.com/RWTH-i5-IDSG/steve-plugsurfing). we have seen that steve is flexible enough to implement such protocols easily and the one for plugsurfing can be taken as inspiration.

rdc-Green, 2022-10-28:
@Safari94 did you manage to implement OCPI with SteVe at all?

juherr, 2022-10-30:
Duplicate of #56