steve-community/steve #487: Build: flyway-maven-plugin fails with MariaDB: Unable to check whether schema `stevedb` is empty

barsnick, 2021-01-26:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.4 or git master HEAD
Operating system  : Fedora 33, Linux 5.10.9
JDK               : OpenJDK 11.0.9.11
Database          : MariaDB 10.4.17
```

#### Expected Behavior

.SteVe builds

#### Actual Behavior

With steve-3.4.4:

```
[INFO] Scanning for projects...
[INFO] 
[INFO] -------------------------< de.rwth.idsg:steve >-------------------------
[INFO] Building steve 3.4.4
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M3:enforce (enforce-java) @ steve ---
[INFO] 
[INFO] --- git-commit-id-plugin:4.0.3:revision (default) @ steve ---
[INFO] 
[INFO] --- properties-maven-plugin:1.0.0:read-project-properties (default) @ steve ---
[INFO] 
[INFO] --- flyway-maven-plugin:7.3.2:migrate (default) @ steve ---
[INFO] Detected a MySQL option file. Upgrade to Flyway Teams Edition to use this for database authentication by visiting https://flywaydb.org/upgrade.
[INFO] Flyway Community Edition 7.3.2 by Redgate
[INFO] Database: jdbc:mysql://localhost:3306/stevedb (MySQL 5.5)
[INFO] Successfully validated 34 migrations (execution time 00:00.187s)
[INFO] Current version of schema `stevedb`: 1.0.2
[WARNING] outOfOrder mode is active. Migration of schema `stevedb` may not be reproducible.
[INFO] Schema `stevedb` is up to date. No migration necessary.
[INFO] Flyway Community Edition 7.3.2 by Redgate
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  5.955 s
[INFO] Finished at: 2021-01-26T08:51:06+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.3.2:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: 
[ERROR] Unable to check whether schema `stevedb` is empty
[ERROR] -------------------------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1728
[ERROR] Message    : Cannot load from mysql.proc. The table is probably corrupted
[ERROR] 
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

#### Steps to Reproduce the Problem

  1. mvn package
[... 137 lines cut ...]
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:957)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:289)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:193)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
[ERROR] 
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
```

barsnick, 2021-03-30:
git bisect says the issue was introduced with this commit:
https://github.com/RWTH-i5-IDSG/steve/commit/288143c26ab158b36a16663380c8c28e4f2f5e57

```
288143c26ab158b36a16663380c8c28e4f2f5e57 is the first bad commit
commit 288143c26ab158b36a16663380c8c28e4f2f5e57
Author: dependabot-preview[bot] <27856297+dependabot-preview[bot]@users.noreply.github.com>
Date:   Mon Sep 28 12:13:28 2020 +0000

    Bump flyway-maven-plugin from 6.5.7 to 7.0.0
[...]
```

goekay, 2021-03-30:
this reads like an issue related to missing privileges to be able to read the db schema. please make sure that you did execute the database steps in https://github.com/RWTH-i5-IDSG/steve#configuration-and-installation

travis can build steve with mysql 5.7 and 8.0 but also with mariadb 10.3 and 10.5: https://travis-ci.org/github/RWTH-i5-IDSG/steve/builds/764649416

barsnick, 2021-03-30:
I used those exact steps from the README when I first created the database.

Here's what MariaDB reports:
```
MariaDB [(none)]> SHOW GRANTS FOR 'steve'@'localhost'
    -> ;
+--------------------------------------------------------------------------------------------------------------+
| Grants for steve@localhost                                                                                   |
+--------------------------------------------------------------------------------------------------------------+
| GRANT USAGE ON *.* TO `steve`@`localhost` IDENTIFIED BY PASSWORD '*****************' |
| GRANT ALL PRIVILEGES ON `stevedb`.* TO `steve`@`localhost`                                                   |
| GRANT SELECT ON `mysql`.`proc` TO `steve`@`localhost`                                                        |
+--------------------------------------------------------------------------------------------------------------+
3 rows in set (0.000 sec)
```

Attempting to use this (which the README doesn't say to do, of course):
```GRANT SUPER ON *.* TO 'steve'@'localhost';```
also has no effect.

goekay, 2021-03-30:
i could not reproduce the error while running `mvn clean package` using the following setup:
- win 7
- jdk 11
- fresh install of mariadb-10.4.18-winx64
- followed steve's instructions to prepare db
- `HEAD` of steve without any local modifications

goekay, 2021-03-30:
i also do not see the following line (from your stack trace):

```
[INFO] Detected a MySQL option file. Upgrade to Flyway Teams Edition to use this for database authentication by visiting https://flywaydb.org/upgrade.
```

therefore, i must assume that you have some local modification which must be breaking the process.

barsnick, 2021-03-30:
Okay, so I just now created a new DB and a new user (stevedb_test_2aa6a783d47d and steve2) along the lines of the README, pointed src/main/resources/config/prod/main.properties to them, but got the same error.

Then I googled around and found out that
```
mysql_upgrade -u root -p
```
does the trick. After this, I was able to build SteVe with the newest flyway plugin. So apparently I was running a new MariaDB with an old scheme? I have no idea. I had created the DB with 10.3.22, and I am at 10.4.18 now.

closed 2021-03-30 by barsnick