steve-community/steve #488: updateType is Full and  localAuthorizationList empty

TarikAkyuz, 2021-01-27:
In Ocpp 1.6, in case of updateType is Full, localAuthorizationList may not be sent. Ocpp 1.6 edition 2.pdf page 69:  

> If no (empty) localAuthorizationList is given and the updateType is Full, all identifications are removed from the
list

Is it possible to simulate this?

lategoodbye, 2021-02-13:
@TarikAkyuz Could you please describe the solution (from UI point of view) you'd like?

lategoodbye, 2021-02-20:
From my side, i see two options here:

1. Extend the select field "update types" according to the following entries
````
   DIFFERENTIAL       -> stays the same
   FULL (Add/Update)  -> sends the complete list
   FULL (Delete)      -> sends a empty list
````
2. Add a new checkbox "Delete list" which is only available for "update types" FULL

goekay, 2021-02-20:
thanks @TarikAkyuz for bringing this detail into attention and @lategoodbye for input. i think this is easily doable. i will look into the options you described and implement one.

goekay, 2021-02-20:
@lategoodbye i'll go with your second option. i do not want to extend the dropdown list, because it currently maps exactly to the UpdateType of ocpp. adding something special for steve would/could create confusion.

goekay, 2021-02-22:
i implemented the change in the referenced commit. please take a look, and re-open the issue if there is an issue (pun intended).

closed 2021-02-22 by goekay