steve-community/steve #489: Confused running project

Momina12, 2021-01-27:
Hi,

I have successfully compiled the project and executed http://127.0.0.1:8080/steve/manager/home. But my central system service page is not showing. Also I don't know how to send reserve request or heart beat request. Can anyone please help me out? 

Kind Regards,

locomoco28, 2021-01-28:
When running SteVe successfully you'll see the endpoint logged into the console.
Remember to run the jar file after compiling.

You can send reservations via the web interface once you get it running. Heartbeats are sent from the charging station so you don't need to send them yourself :)

Momina12, 2021-02-02:
Hi,

Thanks for your help.

I have succesfully executed the central management system page and added dummy charge points o it and sent heartbeat request using SOUPUI. Now, I want to test it with my hardware (charger). What steps should I follow to connect charger with management system?

goekay, 2021-02-05:
> What steps should I follow to connect charger with management system?

this is specific to the charger's configuration management. from the perspective of steve, we only expect

- the chargeBoxId to be added to database such that steve can accept incoming messages using this chargeBoxId, and
- the charger should be configured to use the base SOAP url of steve.

anything besides that is charger territory.

closed 2021-03-22 by goekay