steve-community/steve #491: Remote start transaction without manually filling the form

mathief123, 2021-01-27:
Hi.

First of all, thanks for your amazing job with SteVe ocpp.

Im trying to make a android app that would let me start a charger manually, so, is there a way to fill the remote start transacción form without filling the form manually? I have SteVe installed and it works, but only if i fill the data manually.

Thanks in advance and sorry for my english, as its not my first languaje.

goekay, 2021-02-05:
hello, thanks for the kind words.

steve does not provide an api to control the operations. it only has the web pages.

however, this request pops up from time to time, and there are people who (i think) successfully did stuff like this. i would suggest to search the issues.

goekay, 2021-11-07:
closing this issue due to its inactivity.

closed 2021-11-07 by goekay