steve-community/steve #495: While setting up SteVe on a cloud VM the database migration fails during build

mctsnt, 2021-01-28:
I am trying to create a SteVe instance using a standard Ubuntu VM on DigitalOcean. 
I have seen the wiki about "How to create a SteVe Server using AWS Lightsail" and followed all the steps.
However when I try to build SteVe I get the following error:

`[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.3.2:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException:
[ERROR] Migration V0_6_6__inital.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1419
[ERROR] Message    : You do not have the SUPER privilege and binary logging is enabled (you *might* want to use the less safe log_bin_trust_function_creators variable)`

Since I cannot grant SUPER privileges to the user I was wondering if there is any solution for this?

csamsel, 2021-01-28:
Depends on the actual mysql or mariadb version. But likely no in your case.

mctsnt, 2021-01-29:
I am using mariadb 10.5.8 .
How did you work around this issue in AWS? Would moving the Ubuntu VM I am using to set up SteVe on AWS instead solve this for me?

csamsel, 2021-01-29:
Some MySQL/MariaDB versions do require super rights for stored procedure handling, some dont. Also depending on the service you use, it's actually possibe to reconfigured the mentioned parameter, see: https://aws.amazon.com/premiumsupport/knowledge-center/rds-mysql-functions/

LordDarkneo, 2021-07-30:
Hello,
I am also having this issue, but not on Amazon LightSail, on Docker:

```
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.9.1:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException:
app_1  | [ERROR] Migration V0_6_6__inital.sql failed
app_1  | [ERROR] -----------------------------------
app_1  | [ERROR] SQL State  : HY000
app_1  | [ERROR] Error Code : 1419
app_1  | [ERROR] Message    : You do not have the SUPER privilege and binary logging is enabled (you *might* want to use the less safe log_bin_trust_function_creators variable)
app_1  | [ERROR] Location   : /code/src/main/resources/db/migration/V0_6_6__inital.sql (/code/src/main/resources/db/migration/V0_6_6__inital.sql)
app_1  | [ERROR] Line       : 111
app_1  | [ERROR] Statement  : CREATE TRIGGER `transaction_AINS` AFTER INSERT ON transaction FOR EACH ROW
app_1  | [ERROR]   UPDATE user SET user.inTransaction=1 WHERE user.idTag=NEW.idTag
```

closed 2024-08-07 by goekay