steve-community/steve #496: Steve Build ERROR

freeman-sys, 2021-01-29:
We installed a fresh Ubuntu 20.04.2 LTS and an ubuntu 18.04 with mysql and mariadb both the same problem.
We have tried changing the timezone to 0:0 but not change.

Whe we run: mvn package
we get a lot of WARNING and ERRORs like:
[^[[1;31mERROR^[[m] COMPILATION ERROR :
[^[[1;31mERROR^[[m] /etc/steve/target/generated-sources/jooq/steve/db/tables/ChargeBox.java:[114,78] no suitable method found for createField(org.jooq.Name,org.jooq.DataType<java.time.LocalDateTime>,jooq.steve.db.tables.ChargeBox,java.lang.String,de.rwth.idsg.steve.utils.DateTimeConverter)
[^[[1;31mERROR^[[m] /etc/steve/target/generated-sources/jooq/steve/db/tables/ChargeBox.java:[144,80] no suitable method found for createField(org.jooq.Name,org.jooq.DataType<java.time.LocalDateTime>,jooq.steve.db.tables.ChargeBox,java.lang.String,de.rwth.idsg.steve.utils.DateTimeConverter)

I've attached the log with ERROR and WARNING can someone please tell me what's wrong?
[steve_log.txt](https://github.com/RWTH-i5-IDSG/steve/files/5892386/steve_log.txt)

We run the installation on with the following steps:
Login als root
# apt update
# apt dist-upgrade
# init 6
# apt -y install maven 
# java -version
openjdk version "11.0.9.1" 2020-11-04

# apt install mysql-server
# mysqladmin version
Server version          8.0.22-0ubuntu0.20.04.3
Tried with and withoud mysql_secure_installation
# mysql
CREATE DATABASE stevedb CHARACTER SET utf8 COLLATE utf8_unicode_ci;
CREATE USER 'steve'@'localhost' IDENTIFIED BY 'changeme';
GRANT ALL PRIVILEGES ON stevedb.* TO 'steve'@'localhost';
GRANT SUPER ON *.* TO 'steve'@'localhost';
# init 6
# cd /etc
# git clone https://github.com/RWTH-i5-IDSG/steve.git

Configure SteVe: 
# vi /etc/steve/src/main/resources/config/prod/main.properties
only changed db.password so it's the same as changeme password in SQL statement
# cd /etc/steve
# mvn package > /root/steve_log.txt
Term output:
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release

csamsel, 2021-01-29:
You mean /opt and not /etc? That's not the problem here, but putting something in /etc which doesnt belong there might seriously screw you.
Please try the latest release version to see if it makes a difference. `git checkout steve-3.4.4`

csamsel, 2021-01-29:
fixed in 205060a67286c52fa0fd176a8e6c4fc1f39fa851

freeman-sys, 2021-01-29:
Thank you for the quick reply!
That worked

closed 2021-01-29 by freeman-sys

rdc-Green, 2021-05-16:
hi, we are getting the same problem with 3.4.4on Ubuntu 20.04. I'm not sure what @freeman-sys did to fix the issue?