steve-community/steve #503: Docker build: Log function not available

rybber78, 2021-02-02:
Hi

I'm building and running Steve via Docker successfully. The only thing not working is the log function found in main menu. Pressing the link brings up a blank page saying 'Not available' ?

Any help would be appreciated

goekay, 2021-02-05:
we have two possibilities for logs: to console or to a file. this web page only renders logs if they are written to a file. we have different configs for different environments: https://github.com/RWTH-i5-IDSG/steve/tree/master/src/main/resources/config

logging config can be found here: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/docker/log4j2.xml

as you can see from [line 31](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/docker/log4j2.xml#L31) steve in docker environment logs to console.

docker should have internal mechanisms and config possibilities to handle and route them as desired.

rybber78, 2021-02-06:
Ah, so you mean when using docker the logs will be viewed through eg DOCKER LOGS STEVE_APP command. I'm using this already. But is there a way to debug chargepoint connections, see which data is transmitted RAW, origin IP address etc?

goekay, 2021-02-19:
steve logs incoming and outgoing messages. these should be part of the logs already.

goekay, 2021-03-22:
i consider the questions answered and am therefore closing this issue.

closed 2021-03-22 by goekay

LivingWithHippos, 2023-09-06:
For future readers: copying the log4j2.xml file from prod to docker will enable the log page in docker