steve-community/steve #505: How to build and run this project in eclipse

Droidverine, 2021-02-04:
I'm able to compile and run the project using the commands mentioned in the readme, but I wanted to open the project in eclipse and run it from there. The error issue I'm getting is with the "Flyway".

goekay, 2021-02-05:
sorry, not familiar with eclipse _anymore_, since i use intellij. i switched because of issues like this years ago.

Droidverine, 2021-02-07:
> sorry, not familiar with eclipse _anymore_, since i use intellij. i switched because of issues like this years ago.

Thanks for your response, can you please tell me the steps required to set this project up on intellij?

goekay, 2021-02-07:
the steps are the same as every java project using maven. google is your friend ;)

closed 2021-02-07 by goekay