steve-community/steve #513: OCPP 2.0(.1) Support

andle, 2021-02-19:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
SteVe does not surrently support OCPP 2.0.
...

#### Describe the solution you'd like
OCPP 2.0 support: might have to be a whole new SteVe!?!
...

#### Describe alternative solutions or features you've considered 
Sticking with 1.6 for now :(
...

#### Additional context
I know this would be a huge undertaking, but I just wanted to put the feelers out and see if this is something the SteVe developers have considered. What do you think? Is there something already in progress that I am unaware of?
...

goekay, 2021-02-19:
here is the same question from last summer: https://github.com/RWTH-i5-IDSG/steve/issues/373

there has been no progress in this regard. i will implement ocpp 2.0 some time in the future, but cannot say when.

andle, 2021-02-19:
Thanks for the quick response. And thanks again for SteVe! I'll close this issue.

closed 2021-02-19 by andle