steve-community/steve #514: Question about the content of the error: "has thrown exception, unwinding now" (Probably not a bug)

DeltaVetal26, 2021-02-19:
**Note:** This issue may not be related to the Steve himself. We are trying to figure it out, the only step is to ask you.

In the summer I created a topic where I talked about a script that allows you to remotely send commands to Steve's web control panel.

https://github.com/RWTH-i5-IDSG/steve/issues/338

All is well, this script works completely with Steve on the JSON version of OCPP.

Recently I was approached by a person whose station works on SOAP.

The only thing that could be done with this script and SOAP version is to get the current state of the connector.
All other commands are not processed.
Steve logs contain the following:

[image]

Could you help explain the meaning of this error?
Is it somehow related to the station?

```
SteVe Version     : 3.4.4
Operating system  : Ubuntu
JDK               : 11.0.10
Database          : 5.7.33
```

Thank you!

goekay, 2021-02-19:
hey, 

regarding soap: steve does not know the url of the station to send the message to. `endpoint_address` in table `charge_box` is not set. therefore, this exception happens.

regarding json: there has to be an active/open websocket session to this station. otherwise, you would see an exception with `No session context for chargeBoxId ...`

goekay, 2021-02-19:
more info about `endpoint_address`: every time a station interacts with steve (message flow: station -> steve), one of the headers contains this field. steve will update it, if it is set. apparently it is not.

DeltaVetal26, 2021-02-20:
Hi @goekay,

Thank you for the answer!

Ok, I will try to add an endpoint (soon).
I will report on the result.

goekay, 2021-03-22:
hey @DeltaVetal26, i came across this issue again while i was cleaning up the open issues. what is the status? the problem seems to be about your project that acts as a client to steve. can we close this one?

DeltaVetal26, 2021-05-17:
Hi @goekay, Yes, the issue has been resolved. That was my fault.

Sorry for not answering for so long.

closed 2021-05-17 by DeltaVetal26