steve-community/steve #520: Automate SetChargingProfile

followthebeast, 2021-02-26:
Hello everyone,

First of all, i would like to thank you for the amazing work you're doing.

So i would like to understand if there's any possibility to automate the SetChargingProfile function based in the power of the grid.

I'm still a beginner in this field, i don't have much experience in JAVA web apps either. So i would like to take some recommendations to begin with. I would be grateful if you could help me with any information.

Best regards,

goekay, 2021-02-26:
hey, thanks for the kind words.

> So i would like to understand if there's any possibility to automate the SetChargingProfile function based in the power of the grid.

sure, but "steve on github" (i.e. the main library to build your specific logic on top) will probably not implement this. if you look at the discussion about smart charging in #92 and my comments there, you'll see that i admit that the existing implementation is not optimal, probably not even useful in majority of cases. it can be used for testing and smaller, isolated use cases though. 

in order to make it useful, tighter integration with other apis, systems, smart grid components etc. are necessary. all of this is very very specific for each user/client. therefore the benefits of implementing this for "integration with system X for use case Y" is debatable from my point of view.

followthebeast, 2021-02-26:
Thank you for your reply. I haven't seen this discussion before. I found some interesting stuff on it and it could be helpful for the desired features i want to include.

closed 2021-03-01 by goekay