steve-community/steve #529: Trying to connect a chargepoint via WebSocket and getting exception in SOAP

mctsnt, 2021-03-08:
While trying to connect a chargepoing to SteVe via websocket I am getting the following error message that mentions a SOAP exception. 
In the past I have connected a simulated chargepoint to Steve via websocket with no issues.
I am now trying to connect an actual chargepoint and thus the following error has occured. Does anyone know why this is happening?
```

[ERROR] 2021-03-08 13:50:10,717 de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor - Exception happened
com.ctc.wstx.exc.WstxEOFException: Unexpected EOF in prolog
 at [row,col {unknown-source}]: [1,0]
	at com.ctc.wstx.sr.StreamScanner.throwUnexpectedEOF(StreamScanner.java:701) ~[woodstox-core-6.2.3.jar:6.2.3]
	at com.ctc.wstx.sr.BasicStreamReader.handleEOF(BasicStreamReader.java:2229) ~[woodstox-core-6.2.3.jar:6.2.3]
	at com.ctc.wstx.sr.BasicStreamReader.nextFromProlog(BasicStreamReader.java:2135) ~[woodstox-core-6.2.3.jar:6.2.3]
	at com.ctc.wstx.sr.BasicStreamReader.next(BasicStreamReader.java:1180) ~[woodstox-core-6.2.3.jar:6.2.3]
	at org.apache.cxf.staxutils.DepthXMLStreamReader.next(DepthXMLStreamReader.java:219) ~[cxf-core-3.4.2.jar:3.4.2]
	at org.apache.cxf.staxutils.DepthXMLStreamReader.nextTag(DepthXMLStreamReader.java:231) ~[cxf-core-3.4.2.jar:3.4.2]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:76) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:50) ~[steve.jar:?]
	at org.apache.cxf.phase.PhaseInterceptorChain.doIntercept(PhaseInterceptorChain.java:308) ~[cxf-core-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.ChainInitiationObserver.onMessage(ChainInitiationObserver.java:121) ~[cxf-core-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.http.AbstractHTTPDestination.invoke(AbstractHTTPDestination.java:265) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.ServletController.invokeDestination(ServletController.java:234) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:208) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:160) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.CXFNonSpringServlet.invoke(CXFNonSpringServlet.java:225) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.handleRequest(AbstractHTTPServlet.java:298) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.doGet(AbstractHTTPServlet.java:222) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.service(AbstractHTTPServlet.java:273) ~[cxf-rt-transports-http-3.4.2.jar:3.4.2]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:791) ~[jetty-servlet-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.servlet.ServletHandler$ChainEnd.doFilter(ServletHandler.java:1626) ~[jetty-servlet-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:548) ~[jetty-servlet-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:602) ~[jetty-security-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:235) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1624) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:233) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1435) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:188) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:501) ~[jetty-servlet-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1594) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:186) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1350) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.handle(GzipHandler.java:763) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:59) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.Server.handle(Server.java:516) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.HttpChannel.lambda$handle$1(HttpChannel.java:388) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.HttpChannel.dispatch(HttpChannel.java:633) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:380) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:279) ~[jetty-server-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311) ~[jetty-io-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:105) ~[jetty-io-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.io.ChannelEndPoint$1.run(ChannelEndPoint.java:104) ~[jetty-io-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:336) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:313) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:171) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:129) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:383) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:882) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1036) ~[jetty-util-9.4.38.v20210224.jar:9.4.38.v20210224]
	at java.lang.Thread.run(Thread.java:834) [?:?]
```

goekay, 2021-03-08:
SOAP messages should not be used with websocket. either SOAP (with default http) or JSON (via websockets). i think this might be configuration mistake.

mctsnt, 2021-03-08:
@goekay Thank you for replying! Yeah that was why I was surprised to see the error in SOAP.
Do you mean a charging station configuration issue or something with Steve? I haven't really changed anything in SteVe when it comes to the configuration.

goekay, 2021-03-08:
i mean charging station configuration. 

i can imagine the following: it is a web socket & json station, but it was configured to use the soap url. therefore, it tries to initiate the connection and sends some data packets (websocket upgrade and handshake frames maybe?). but this soap endpoint was expecting a proper xml message. and hence the error.

csamsel, 2021-03-08:
Yes make sure you enter the SOAP endpoint of SteVe into the charging station if it is using SOAP (ocpp1.x**S**), and the Websocket endpoint if it's configured for JSON (ocpp1.x**J**)

closed 2021-04-23 by mctsnt