steve-community/steve #532: J1772 protocol, ocpp and state of charge (SOC)....

retrocoder-78, 2021-03-13:
Hello Steve team!

In my adventures in understanding ocpp and in relation to the j1772 standard, I have come across state of charge (SOC) in the ocpp documentation. I have read all the available information and the ocpp protocol 1.6J looks as if it does have a mechanism for defining state of charge as a percentage digitally via power line communication.

I guess my question, has the Steve team investigated at all this state of charge functionality of the ocpp protocol? To have access to a vehicle's state of charge would add value to the metering that is occurring during a charge session.

Any information anyone has on this topic would be much appreciated!

goekay, 2021-03-15:
ocpp defines a MeterValues message for the charging station to send a variety of measurements. SoC is one of them (i.e. any MeterValue with `Measurand.SoC` as the measurand). so, ocpp supports this. steve supports this as well (i.e. we write every MeterValue into db). however, we do not have any logic to do something with these values.

retrocoder-78, 2021-03-15:
Thanks for the information goekay. I have watched the metered data coming from a charge session currently with a 10-second time stamp, v, a, wh. Which is so valuable to track energy usage. So, what you're saying is during a charge session this request for data could be made and a 1.6J compliant charging station in theory should report the request as a percentage as per the ocpp specification.

Would there be any benefit or value to Steve making this request and reporting it alongside the aforementioned information? Maybe I missed it, but soc is currently not being requested by Steve to report on during a charge session. Do I have that right? Along with the 10-second interval for reporting wh, a, v, could soc (ocpp specified %) be added?

goekay, 2021-03-16:
> a 1.6J compliant charging station in theory should report the request as a percentage as per the ocpp specification.

not _should_ but _could_. 

first of all, the charging station should support and implement this feature. you can send a `GetConfiguration` request to the station to see whether the station supports `SoC`.

then, you can send a `ChangeConfiguration` request from steve with the key `MeterValuesSampledData` and as value a comma separated list with things you can the station to report. this list can contain `SoC`. with this steve is basically telling the station "please send me SoC as part of the MeterValues". with the same `ChangeConfiguration` you can configure the time interval for MeterValues. you should look into the MeterValues and ChangeConfiguration parts of the ocpp spec for more info around these features.

retrocoder-78, 2021-03-16:
I couldn't agree more @goekay. I am finding as I move from charge station to charge station the implementation of the entire 1.6J protocal varies from unit to unit, oem to oem. Which is really quite frustrating as the protocal is so well defined. But, is what it is.

Thank you for the detailed information. Much appreciated. I am going to dig into what you have listed and the protocol further. I guess in an ideal world steve would report the soc as the current meter values are coming in while a charge station is occurring. Or, at some predetermined interval. As the vehicle is consuming energy, wh, v, a, the soc will be changing in proportion to the size of the battery pack (khw) and a function of the charge rate (total wattage). But, I can see why SoC is not implemented in steve as would all charge stations support such. Wh, v, a, absolutely. SoC, I don't think it's as clear cut as the variables that steve does actively report on now. 

The information you have provided is really super helpful! To me SoC is a really important piece to a charge session whether it be L1, L2, or L3. To this end, this is why DC fast chargers both Chademo and CCS all must report SoC as the 'charger' is outside the vehicle bypassing internal equipment. At the end of the day, electric vehicles and their state of charge are a piece of the larger energy picture. I hope to report back soon with some findings!

goekay, 2021-03-16:
> ... I guess in an ideal world steve would report the soc as the current meter values are coming ...

> ... But, I can see why SoC is not implemented in steve as would all charge stations support such ...

> ... SoC, I don't think it's as clear cut as the variables that steve does actively report on now ...

what do you mean by these? steve is not the decider. charging station is. steve does not report anything. if the station sends/reports the soc within MeterValues to steve, steve will write it into database. by default, steve does nothing with this data afterwards. but the data will be there. there is no loss. you can do anything with that. you can query it, do your analysis based on it.

goekay, 2021-03-16:
maybe your definition of reporting is a different one. if you mean by "steve does not report the soc" that "steve does not present this soc information on web pages", i agree.

retrocoder-78, 2021-03-16:
> what do you mean by these? steve is not the decider. charging station is. steve does not report anything. if the station sends/reports the soc within MeterValues to steve, steve will write it into [...]

Oh I see @goekay. Okay. I think I am understanding this. Maybe I'll rephrase my wording. So, just to further clarify, if SoC is being sent from the charge box then it won't be visible along side wh, v, and a on the Steve web page currently. Regardless if visible or not, it would be logged in the database. Which is great! Data received is what's important. 

Just saw your other reply. I think we're saying the same thing. 

Again, I really appreciate you time and information.

goekay, 2021-03-16:
> if SoC is being sent from the charge box then it won't be visible along side wh, v, and a on the Steve web page currently. Regardless if visible or not, it would be logged in the database.

exactly... with a nuance regarding its visibility. just looked at this part of the code again and refreshed my memory. i need to slightly correct my answer: _we do not take actions to specifically disable soc on the web page._

if you go to the "transaction details" page of a transaction id, you will see a table with a list of MeterValue rows. this table blindly renders _all_ MeterValue data associated with this transaction id. if station sent soc data, it will be there. if station did not send soc data, it will be not there.

JeremyWhaling, 2021-04-14:
Standard J1772 AC does not support SoC. The communication from the vehicle is only "vehicle is requesting charging" or "not requesting". The station only provides the ampacity available on a PWM signal on the pilot wire. There's also "station ready" "EV ready" and "error" states, but that's basically it. It's a very simple protocol. Too simple really. Shame SOC wasn't considered important when it was created.

You'd need to have ISO 15118 support on the vehicle (or use DC charging via CCS) in order to get SoC. Only a handful of new vehicles support ISO 15118, unfortunately.

goekay, 2021-05-21:
i consider this issue resolved and am closing it.

closed 2021-05-21 by goekay

elad-eyal, 2023-03-26:
@goekay quick question if you can 

> first of all, the charging station should support and implement this feature. you can send a `GetConfiguration` request to the station to see whether the station supports `SoC`.

Which configuration item will tell me if my station "Supports SoC" ? Is it part of SupportedFeatureProfiles ?

I tried sending a `ChangeConfiguration` request from steve with the key `MeterValuesSampledData` including `SoC` and the request was "Accepted" but I did not see any SoC information. The station is Wallbox Pulsar Plus.

shadsnz, 2023-03-26:
Wallbox chargers do not provide the SoC information in our experience. The distributor advised us they may introduce it in the future.