steve-community/steve #533: Maven compilation failed

followthebeast, 2021-03-16:
#### Specifications
```
SteVe Version     : 3.44
Operating system  : Windows 10
JDK               : 11.0.10
Database          : MySql 8.0.23
```

#### Actual Behavior

Hello, when i tried to compile Steve in powershell with the command mvn package, i got this error :

[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] /C:/Users/Jihed/Pictures/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[112,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>
[ERROR] /C:/Users/Jihed/Pictures/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[113,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>
[INFO] 2 errors
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:11 min
[INFO] Finished at: 2021-03-16T09:46:26+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.1:compile (default-compile) on project steve: Compilation failure: Compilation failure:
[ERROR] /C:/Users/Jihed/Pictures/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[112,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>
[ERROR] /C:/Users/Jihed/Pictures/steve-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[113,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

OSkrk, 2021-03-17:
Same problem here !

[ERROR] COMPILATION ERROR : [INFO] -------------------------------------------------------------[ERROR] /home/ubuntu/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[112,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>[ERROR] /home/ubuntu/steve/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[113,58] incompatible types: org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Integer> cannot be converted to org.jooq.TableField<jooq.steve.db.tables.records.OcppTagActivityRecord,java.lang.Boolean>[INFO] 2 errors

ShortFust, 2021-03-17:
Same problem here

using JDK 11.0.10 on ubuntu
the github clone generate this error,
but when using lastest release (steve-3.4.4 on 30 Dec 2020) it works fine.

but the github clone work on my local environment (JDK 11.0.2 on windows 10)

goekay, 2021-03-17:
are you all using mysql 8?

followthebeast, 2021-03-17:
> are you all using mysql 8?

For my case, yes ! I tried to roll back the pom.xml file and now it works. I think you must verify the latest update @goekay .

Have a good day !

OSkrk, 2021-03-17:
What do you mean by roll back the pom.xml ?

goekay, 2021-03-17:
@followthebeast reported that 3.4.4 has this issue. 
@ShortFust reports that the HEAD of the repo has this issue and using 3.4.4 fixed it.

the two reports express different facts :)

goekay, 2021-03-19:
i could reproduce the issue with mysql 8 and the HEAD of the repo. it was caused by removal of [these lines](https://github.com/RWTH-i5-IDSG/steve/commit/f5e78706aaecb4f5c33559030b99cb8b12e0ae84#diff-9c5fb3d1b7e3b0f54bc5c4182965c4fe1f9023d449017cece3005d3f90e8e4d8L400-L401) during a clean-up and modification of pom.xml. i was under the impression these were redundant because the [new expression](https://github.com/RWTH-i5-IDSG/steve/commit/f5e78706aaecb4f5c33559030b99cb8b12e0ae84#diff-9c5fb3d1b7e3b0f54bc5c4182965c4fe1f9023d449017cece3005d3f90e8e4d8R397) would cover these as well. 

apparently, this worked fine for only mysql 5.7.x but not for mysql 8. since the travis jobs did not cover mysql 8, i was not aware of it. @csamsel is it possible to add mysql 8 to the job matrix?

my referenced commit above fixes this problem by re-introducing removed configuration. sorry for the caused trouble.

csamsel, 2021-03-19:
I don't think mysql 8  made into Ubuntu yet, hence it wasn't covered. Personally I've steered more to MariaDB which seems to be more robust. I'll add a job for mysql 8 but we should consider recommending MariaDB 10
3 as default.

csamsel, 2021-03-20:
@goekay i made in improve pass on the travis config. the jobs with MySQL 8.0 fail indeed

goekay, 2021-03-20:
thanks, i am merging the branch above to fix builds of master.

goekay, 2021-03-20:
since we solved the problem, i am closing this issue. thanks to all participants for the reports and help.

closed 2021-03-20 by goekay