steve-community/steve #549: How do I connect EVBox (22 kW) to SteVe?

ashupani10, 2021-04-06:
Hi, 

I want to connect my EVBox to the SteVe backend. I followed the instructions as provided in the Github page.

For EVBox, I went to EVBox connect app --> Installer Mode --> Charging Management Platform --> Different backend URL --> ws://<server address for SteVe>/steve/websocket/CentralSystemService/<chargebox ID>/

However, I am not able to get a heartbeat on SteVe.

Could someone please help me out with this.

Thanks in advance

goekay, 2021-04-06:
you probably do not have to add the chargebox id yourself. the station might add it itself. therefore, the url can be like: 
ws://<..server address for SteVe..>/steve/websocket/CentralSystemService/

ashupani10, 2021-04-06:
Thanks, 

I will try it out. What should I expect to happen? Should the EVbox just pop up on SteVe?

goekay, 2021-04-06:
as a starter, you should see the station under "number of connected charge points". additionally, "received heartbeats" and "connector status" pages should list the station (and its connectors) as well.

ashupani10, 2021-04-07:
Hi, 

Thank you for the reply, however, the charger is not seen on SteVe. Could any of the following cause the problem?

1.  EVBox Elvi (22kW) might have updated firmware that is not supported by SteVe or vice-versa. (But this should not be the case right? as OCPP is an open protocol)
2. So my test setup is just a powered EVBox, there is no connector or EV connected - see image attachment. Could that cause an issue?
[image]

followthebeast, 2021-04-07:
@ashupani10 when you compile steve, you're just creating an internal web server in your machine (your server address for SteVe. : localhost). So to connect your EVBox Elvi, you must create a network environment where your charging station can communicate with steve web server. 
In short, I think that your setup is not good. I can tell you that i have the same issue with EVBox Elvi model because it only connects in wifi or 4g. Steve is working properly for charging station with ethernet connection if you have a local solution.

ashupani10, 2021-04-07:
Hi @followthebeast ,

Thank you for your reply. 
That is indeed something that I will work on. As you might have guessed, I am pretty new to all this stuff. It would be great if I could get some guidance (tutorials, sources) to set up such a network environment to connect the EVBox. I will also search on my end how to do the same. 

Meanwhile, I had some other questions related to SteVe.

1.  I want to implement smart charging and dynamic power management algorithms through SteVe, Is it possible to implement such smart charging algorithms?
2. If yes, what do think would be the response time of such a system? 10 - 15 seconds, less, more?

Thank you again for your continual support,

Warm regards,

Ashutosh

goekay, 2021-04-07:
the issue looks like to be related to the station. unfortunately, i cannot help further in this regard.

> I want to implement smart charging and dynamic power management algorithms through SteVe, Is it possible to implement such smart charging algorithms?

yes, ocpp supports smart charging. and steve implements this part of ocpp. 

> If yes, what do think would be the response time of such a system? 10 - 15 seconds, less, more?

response time of what? this question is too vague. it all depends on your context.

ashupani10, 2021-04-08:
@goekay, by response time what I mean is the following: 

Suppose the car is already charging with a current of 14 amps and I want to reduce it to 10 A. I send a command through Steve. What do you think would be the time between sending the charging signal through Steve and actual drop of current in the physical system?

csamsel, 2021-04-08:
> @ashupani10 when you compile steve, you're just creating an internal web server in your machine (your server address for SteVe. : localhost). So to connect your EVBox Elvi, you must create a network [...]

There is no reason why WiFi or 4G would not work with SteVe. It's like you said, a working network connection between the station and the Steve application is required. If that's a problem, i suggest asking someone with expertise in that area. Done wrong it could lead to security issues.

followthebeast, 2021-04-08:
> There is no reason why WiFi or 4G would not work with SteVe. It's like you said, a working network connection between the station and the Steve application is required. If that's a problem, i [...]

Yes, i'm talking just about my little experience with steve. Tested with a charging station locally and it works perfectly fine.

> @goekay, by response time what I mean is the following:
>
> Suppose the car is already charging with a current of 14 amps and I want to reduce it to 10 A. I send a command through Steve. What do you [...]

I think it depends of the charging station on itself. For example, i have tested a Schneider Smart Wallbox, to clear and set a charging profile, it takes around 5 to 7 seconds.

ashupani10, 2021-04-08:
Hey Guys!

Thanks a lot! I got it connected!!

goekay, 2021-04-09:
@ashupani10 i am closing this issue, since i consider it resolved after your comment.

closed 2021-04-09 by goekay

axen2u, 2022-06-08:
@jpaquete  Check your URL Normally It should be like this if you're using JSON websocket 
`ws://<ip>:<port>/steve/websocket/CentralSystemService/<Chargeboxid>`

I found some boxes automatically adding (ex:- node-red-contrib-ocpp) chargerbox id at the end of the URL. In those cases just use 
`ws://<ip>:<port>/steve/websocket/CentralSystemService`

wlanrouter, 2023-09-06:
> Hey Guys!
> 
> Thanks a lot! I got it connected!!

Could you please describe your settings? What is the APN setting on the wallbox? What is platform management?
What is the exact websocket address at the wallbox? With WallboxId at the end or not?

What must be done on SteVe to get a heartbeat?