steve-community/steve #560: Where to add a query?

rizukijr, 2021-04-17:
I want to add a single input to the column I just created, along with data generation in the transaction_stop table, where can I find the file that runs the append data query in the transaction_stop table?

goekay, 2021-04-18:
the insertion of the data happens here: https://github.com/RWTH-i5-IDSG/steve/blob/364ae967d0a7bafd7da8838a65a4d13dc9918dfd/src/main/java/de/rwth/idsg/steve/service/CentralSystemService16_Service.java#L194-L223

other uses interact with: https://github.com/RWTH-i5-IDSG/steve/blob/364ae967d0a7bafd7da8838a65a4d13dc9918dfd/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java

please follow the code flows from and to these classes in order to realize what you want.

rizukijr, 2021-04-19:
thank you very much for the information

closed 2021-04-19 by rizukijr