steve-community/steve #563: Clean connector_meter_value Table after a certain period of time

followthebeast, 2021-04-22:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

Hello,

I want to know does steve clean DB after a certain period of time ? 

For example, connector_meter_value size on the disk can be large after using steve in many transactions.

[image]

Best regards,

goekay, 2021-04-22:
no, we do not. it is an operational decision based on the business cases. therefore, users of steve should do it themselves... if ever. btw 3k rows are nothing for a production system, unless you have tight hardware requirements (e.g. raspberry pi).

followthebeast, 2021-04-22:
Thank you @goekay for your answer. Yes 432 kib is nothing but as i told you, i think at a certain time it will be more and more oversized and it will take time to make operations. 

One last question, you think adding a db migration file with this sql statement : 

delete from connector_meter_value where value_timestamp < DATE_SUB(NOW() , INTERVAL 1 MONTH)

could work to delete data before one month?

goekay, 2021-04-22:
two things:

1. technical aspect: if you add this query to a migration file, it will be executed only once (during the execution of the migration file). you want to run this kinda query every day/week/month/etc. you actually need a cron job that executes it. therefore, this approach does not give you the solution that you want.
2. business aspect: how much do you care about your historical data? do you do something with your historical data? what do you do with your historical data? this table includes, for example, intermediate values of a transaction between start and stop. you might not want to delete this info... ever. different types of data in this table might have different priorities to you. therefore, you probably should not "paint it with the same brush" by deleting everything older than X days/weeks/months/etc. if you do not care for it, go for it.

followthebeast, 2021-04-22:
Thanks @goekay . 
Yes you're right i think i need a cron job that executes it. And as you said some of this data is important like the start/stop transaction value but for my usercase i will only delete data from connector_meter_value table. the start/stop transaction meter values will be always stored in their corresponding tables.

goekay, 2021-04-22:
consider the following actions for a transaction flow:

- a transaction X was started at 10:00 am with meter value of 100 (will be in transaction_start table)
- the transaction X is at 10:10 am at meter value of 150 (will be in connector_meter_value table)
- the transaction X is at 10:20 am at meter value of 200 (will be in connector_meter_value table)
- the transaction X is at 10:30 am at meter value of 250 (will be in connector_meter_value table)
- the transaction X is at 10:50 am at meter value of 300 (will be in connector_meter_value table)
- the transaction X was stopped at 11:00 am with meter value of 320 (will be in transaction_stop table)

the intermediate meter values of a transaction will be in connector_meter_value table.

followthebeast, 2021-04-22:
Exactly 👍. Thanks for your precious time @goekay . I consider this issues is solved and it can be closed.

closed 2021-04-22 by goekay