steve-community/steve #564: Get Composite Schedule request keeps getting rejected

mctsnt, 2021-04-23:
Hello! I have successfully connected a charging point to SteVe and I am able to do every OCPP 1.6 operation without issue including setting up a charging profile. However when I try to "Get Composite Schedule" the response I always get is Rejected. Does anyone know why that could be happening?

The charging schedule I have set is this:

Charging Profile Purpose: TxDefaultProfile
Charging Profile Kind: ABSOLUTE
Recurrency Kind: DAILY
Duration (in seconds): 86400
Charging Rate Unit: A

Start Period (in sec) = 0, Power Limit (in amperes) = 16, Number Phases = 1

And the request parameters for the "Get Composite Schedule" I set are these:
Connector ID: 1 (i set the profile for connector 1 but I have also tried it with 0 and nothing changed)
Duration (in seconds): 86400
Charging Rate Unit: A

chuck-h, 2021-04-23:
FYI I just checked that this works ok for me with a Delta AC Mini. Here is the log, you might compare it to yours.

```
[INFO ] 2021-04-23 08:54:50,153 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Delta01, sessionId=d7de37e] Sending: [2,"7bc80158-7d07-452a-9380-725d833f6ccb","GetCompositeSchedule",{"connectorId":1,"duration":100,"chargingRateUnit":"A"}]
[INFO ] 2021-04-23 08:54:51,578 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=Delta01, sessionId=d7de37e] Received: [3,"7bc80158-7d07-452a-9380-725d833f6ccb",{"status":"Accepted","connectorId":1,"scheduleStart":"2021-04-23T15:54:53Z","chargingSchedule":{"duration":100,"startSchedule":"2021-04-23T15:54:53Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":12.0,"numberPhases":1}],"minChargingRate":0.0}}]
```
I am running an older build of SteVe, however.

lategoodbye, 2021-04-24:
@mctsnt What makes you sure that the charging point supports "Get Composite Schedule"?

mctsnt, 2021-04-24:
> @mctsnt What makes you sure that the charging point supports "Get Composite Schedule"?
There is nothing in the Get Configuration for example to indicate that it doesn't. Also it supports all other OCPP 1.6 operations so far I don't see how this would be any different. (The station is an EVBox Elvi btw)

lategoodbye, 2021-04-24:
Nevertheless you should should ask EVBox support about this behavior, because SteVe sends a valid request to the charging point and there is no chance that the authors of SteVe can say anything about the closed charging point software.

goekay, 2021-04-24:
"Get Composite Schedule" is a simple message. there is not much Steve could do wrong while setting the parameters and sending it.  

> I am running an older build of SteVe, however.

we did not touch its implementation for a long time. if it was running before, it should be running now.

@mctsnt if you have access to the logs of the station, please take a look at them. maybe you can find something there.

mctsnt, 2021-04-24:
@goekay  Can't really see anything helpful from the log so far:

`[INFO ] 2021-04-24 15:27:45,283 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=EVB-P20210549, sessionId=80ac3314-d775-613c-b917-a513feba6e2e] Sending: [2,"1d205798-640d-497c-9e36-ed4370ea893f","GetCompositeSchedule",{"connectorId":1,"duration":100,"chargingRateUnit":"A"}]
[INFO ] 2021-04-24 15:27:45,504 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=EVB-P20210549, sessionId=80ac3314-d775-613c-b917-a513feba6e2e] Received: [3,"1d205798-640d-497c-9e36-ed4370ea893f",{"status":"Rejected"}]`

Anyways since this is probably not a SteVe problem I'll close the issue and try EVBOX for help. Thanks for all the replies everyone!

closed 2021-04-24 by mctsnt

goekay, 2021-04-24:
these logs you posted are from Steve. i meant loga of the charging station.

mctsnt, 2021-04-24:
Oh sorry I misunderstood, I'll check it out. Once again thanks for the help!