steve-community/steve #571: Retrospectively add transaction id

rdc-Green, 2021-04-27:
I have a few transactions that have been created without a transaction_pk. Is it possible to manually add a transaction_pk to them. I tried this and it said it couldn't update due to no matching foreign key.

goekay, 2021-04-29:
this is not possible yet. can you please be more specific about the message flow?

transaction_pk is determined by the database after a StartTransaction from the station arrives. StopTransaction should then reference this transaction_pk. i assume, the station was working in offline mode and did not contact steve for StartTransaction? therefore, for both StartTransaction and StopTransaction a dummy one was used?

rdc-Green, 2021-04-29:
Thanks @goekay that makes sense. I think i'll just remove these transactions as all is working fine now and has been for a while. Just wanted to see if there was an easy way to edit them.

goekay, 2021-04-29:
adding edit functionality is easy. editing them _while ensuring data integrity_ is hard.

the rows in transaction_start and transaction_stop tables are linked with transaction_pks. imagine lots of entries with transaction_pk `-1`  in transaction_start and transaction_stop: _which -1 entry in transaction_start should be associated with which -1 entry transaction_stop?_

this is tricky to answer and implement. it would involve lots of assumptions. on the other side, adding a simple edit functionality and giving the responsibility to the user would enable the user to break things.

rdc-Green, 2021-04-29:
I would definitely break things!

goekay, 2021-04-29:
> I have a few transactions that have been created without a transaction_pk.

i still cannot understand what you mean by this. transaction_pk cannot be null. there is a not null constraint on both tables in db.

rdc-Green, 2021-04-29:
See image below of transaction_pk with 'NULL' value

[image]

goekay, 2021-04-29:
this is neither transaction_start nor transaction_stop. this is the connector_meter_value table. it is a completely different beast. 

it does not mean that there are transactions without transaction_pks.

a station can send any sensory data to steve. these are stored in connector_meter_value. SOME of these sensor data might be related to a transaction. in these cases, transaction_pk will be set. in other cases, it will be null. this is okay.

rdc-Green, 2021-04-29:
oh ok. Can I add a transaction_pk to this table though. I can see the missing transaction_pks from the transaction_start table. Its between 1064 and 1067

[image]

goekay, 2021-04-29:
> Can I add a transaction_pk to this table though. I can see the missing transaction_pks from the transaction_start table. Its between 1064 and 1067

what you are talking about is not setting a transaction_pk for a transaction. what you are talking about is filling a gap in the history, and adding (lost?) transactions (aka. charging sessions) about which steve knows nothing. how do you want to add specifically? do you know the charging station? its connector? the idtag? start timestamp? and start value? manually adding a lost transaction is error prone. i would advise to look into why these were lost, and fix the underlying issue. then, you will not have a gap in your transaction history.

as a general note:

please use the issue template and give as much as possible information at the beginning for me to help you more efficiently. not seeing the screenshot led me to a completely different path while trying to analyze the issue.

closed 2021-04-29 by goekay

rdc-Green, 2021-04-29:
Sorry @goekay i'll do that next time. I do know all the details you mention above as they are in the connector_meter_value table