steve-community/steve #572: Sending Charging Profile with A or W limit

Dan-27, 2021-04-28:
Hi, I would like to set a charging profile in SteVe. How can I set the limit of Amps or Watts in the charging profile?

closed 2021-04-29 by Dan-27

hackex, 2021-06-12:
Has this been answered?
I would also like to know how I could, for example, realize PV surplus charging through setting of Amp limits (sending the needed Amp per Phase to the wallbox)

Thanks to anyone who has information on that

goekay, 2021-06-12:
@hackex you can define charging profiles under data management section of steve. these charging profiles can use A or W. then, you can send a charging profile to a station using the corresponding operation under operations menu.

hackex, 2021-06-12:
> @hackex you can define charging profiles under data management section of steve. these charging profiles can use A or W. then, you can send a charging profile to a station using the corresponding operation under operations menu.

thanks for the reply. Can I also control these values based on some external real-time values, e.g. PV generation? Or how do I get external data into SteVe?

goekay, 2021-06-12:
> Can I also control these values based on some external real-time values ...
> Or how do I get external data into SteVe?

you cannot. please browse the issues for more context and reasoning.

> Does sending a charging profile can only be done over the menu or is there some API?

there is no API.