steve-community/steve #578: Question

pulkitgarg007, 2021-05-15:
Can anyone help me to understand where and how these endpoints are configured.
SOAP: http://<your-server-ip>:<port>/steve/services/CentralSystemService
WebSocket/JSON: ws://<your-server-ip>:<port>/steve/websocket/CentralSystemService/<chargeBoxId>

Thanks for your help.

csamsel, 2021-05-16:
Depends what you want to achieve

goekay, 2021-05-17:
soap: `services/CentralSystemService` is just a dummy endpoint and mediator to forward requests to the correct implementation wrt. ocpp 1.2, 1.5 and 1.6. 

https://github.com/RWTH-i5-IDSG/steve/blob/7dd2d78e3666b46dbe077cb18bd1f9e09395cfe0/src/main/java/de/rwth/idsg/steve/config/OcppConfiguration.java#L70-L81

websocket: similarly `websocket/CentralSystemService` is the handshake endpoint for different ocpp-j version implementations. the actual implementation will be chosen during handshake based on the websocket subprotocol that references the ocpp version.

https://github.com/RWTH-i5-IDSG/steve/blob/7dd2d78e3666b46dbe077cb18bd1f9e09395cfe0/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L69-L91

take a look at the injected objects used in the snippets above and please follow the rabbit.

closed 2021-05-21 by goekay

pulkitgarg007, 2021-05-22:
Thanks Sevket. It helped