steve-community/steve #579: MySQL database migration to new VM

rdc-Green, 2021-05-16:
We are trying to migrate Steve to a new server. We have successfully created steve 3.4.4 with a new MySQL database, but cannot connect to the old database. We have added new tables to the old database, but not sure if that is what is causing the problem. 

Both the db and steve are using the same time zone. We have migrated from Ubuntu 18.04 to Ubuntu 20.04 (although I don't think that is the issue as we managed to compile steve using new MySQL db, just not using the old one.

Please see errors below:

`[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.1:compile (default-compile) on project steve: Compilation failure: Compilation failure:
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/service/OcppTagService.java:[96,77] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/service/OcppTagService.java:[123,40] incompatible types: java.lang.Integer cannot be converted to @org.jetbrains.annotations.Nullable org.joda.time.DateTime
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/service/OcppTagService.java:[178,47] incompatible types: java.lang.Integer cannot be converted to org.joda.time.DateTime
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/service/OcppTagService.java:[209,69] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/service/OcppTagService.java:[213,97] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/web/controller/OcppTagsController.java:[94,38] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/web/controller/OcppTagsController.java:[96,51] incompatible types: java.lang.Integer cannot be converted to org.joda.time.DateTime
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/web/controller/OcppTagsController.java:[102,36] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/web/controller/OcppTagsController.java:[104,51] incompatible types: java.lang.Integer cannot be converted to java.lang.String
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[271,54] no suitable method found for eq(org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer>)
[ERROR]     method org.jooq.Field.eq(java.lang.String) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to java.lang.String)
[ERROR]     method org.jooq.Field.eq(org.jooq.Field<java.lang.String>) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.jooq.Field<java.lang.String>)
[ERROR]     method org.jooq.Field.eq(org.jooq.Select<? extends org.jooq.Record1<java.lang.String>>) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.jooq.Select<? extends org.jooq.Record1<java.lang.String>>)
[ERROR]     method org.jooq.Field.eq(org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.String>>) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.String>>)
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[301,57] no suitable method found for eq(java.lang.String)
[ERROR]     method org.jooq.Field.eq(java.lang.Integer) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to java.lang.Integer)
[ERROR]     method org.jooq.Field.eq(org.jooq.Field<java.lang.Integer>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.Field<java.lang.Integer>)
[ERROR]     method org.jooq.Field.eq(org.jooq.Select<? extends org.jooq.Record1<java.lang.Integer>>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.Select<? extends org.jooq.Record1<java.lang.Integer>>)
[ERROR]     method org.jooq.Field.eq(org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.Integer>>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.Integer>>)
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[320,25] no suitable method found for date(org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer>)
[ERROR]     method de.rwth.idsg.steve.utils.CustomDSL.date(org.jooq.Field<org.joda.time.DateTime>) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.jooq.Field<org.joda.time.DateTime>)
[ERROR]     method de.rwth.idsg.steve.utils.CustomDSL.date(org.joda.time.DateTime) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.joda.time.DateTime)
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[329,25] no suitable method found for date(org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer>)
[ERROR]     method de.rwth.idsg.steve.utils.CustomDSL.date(org.jooq.Field<org.joda.time.DateTime>) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.jooq.Field<org.joda.time.DateTime>)
[ERROR]     method de.rwth.idsg.steve.utils.CustomDSL.date(org.joda.time.DateTime) is not applicable
[ERROR]       (argument mismatch; org.jooq.TableField<jooq.steve.db.tables.records.TransactionRecord,java.lang.Integer> cannot be converted to org.joda.time.DateTime)
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[341,52] no suitable method found for between(org.joda.time.DateTime,org.joda.time.DateTime)
[ERROR]     method org.jooq.Field.between(java.lang.Integer,java.lang.Integer) is not applicable
[ERROR]       (argument mismatch; org.joda.time.DateTime cannot be converted to java.lang.Integer)
[ERROR]     method org.jooq.Field.between(org.jooq.Field<java.lang.Integer>,org.jooq.Field<java.lang.Integer>) is not applicable
[ERROR]       (argument mismatch; org.joda.time.DateTime cannot be converted to org.jooq.Field<java.lang.Integer>)
[ERROR] /home/greenpointev/greenpointev-master/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[87,63] no suitable method found for eq(java.lang.String)
[ERROR]     method org.jooq.Field.eq(java.lang.Integer) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to java.lang.Integer)
[ERROR]     method org.jooq.Field.eq(org.jooq.Field<java.lang.Integer>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.Field<java.lang.Integer>)
[ERROR]     method org.jooq.Field.eq(org.jooq.Select<? extends org.jooq.Record1<java.lang.Integer>>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.Select<? extends org.jooq.Record1<java.lang.Integer>>)
[ERROR]     method org.jooq.Field.eq(org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.Integer>>) is not applicable
[ERROR]       (argument mismatch; java.lang.String cannot be converted to org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.Integer>>)
[... cut ...]
[ERROR]       (argument mismatch; org.joda.time.DateTime cannot be converted to org.jooq.QuantifiedSelect<? extends org.jooq.Record1<java.lang.Integer>>)
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException`

csamsel, 2021-05-16:
Hard to say what's going wrong, espacially when you made modifications. A backup and restore using the method here:
https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-can-i-backup--transfer-the-steve-database
with a new compilation should work. Based on the error message it could be an incompatible jooq version or configuration.

rdc-Green, 2021-05-16:
thank you @csamsel we followed your instructions. When we first try to migrate we migrated multiple dbs at once. When following your instructions we just did it with a single db. Its all working now on Ubuntu 20.04

csamsel, 2021-05-16:
Happy to help, you probably missed the correct export options in the first export.

closed 2021-05-26 by rdc-Green