steve-community/steve #581: Exception adding new charging profile

barsnick, 2021-05-20:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : git master 7dd2d78e3666b46dbe077cb18bd1f9e09395cfe0
Operating system  : Linux 5.11.20, Fedora 33 (kernel-5.11.20-200.fc33.x86_64), running inside VirtualBox
JDK               : OpenJDK 11.0.11.0.9 (java-11-openjdk*-11.0.11.0.9-2.fc33.x86_64)
Database          : MariaDB 10.4.19
```

#### Expected Behavior

Charging profile is added, or an error message is presented.

#### Actual Behavior

Backtrace

#### Steps to Reproduce the Problem

  1. In SteVe's web interface, select Data Management -> Charging Profiles -> Add New
  2.1. Description: bla
  2.2. Stack level: 0
  2.3. Charging Profile Purpose: CHARGE_POINT_MAX_PROFILE
  2.4. Charging Profile Kind: RELATIVE
  2.5. Charging Rate Unit: A
  3. "Add" (Schedule Periods)
  3.1. Start Period: 0
  3.2. Power Limit: 22.3
  3.3. Number Phases: empty
  4. Press the blue/green "Add" button

#### Additional context
<details>
<summary>Here's the complete backtrace:</summary>

```
[ERROR] 2021-05-20 18:56:22,148 org.springframework.web.servlet.tags.form.InputTag - Invalid property 'schedulePeriodMap[zz1621529663529].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1621529663529].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?
org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1621529663529].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1621529663529].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?
	at org.springframework.beans.AbstractNestablePropertyAccessor.getPropertyValue(AbstractNestablePropertyAccessor.java:627) ~[spring-beans-5.3.7.jar:5.3.7]
	at org.springframework.beans.AbstractNestablePropertyAccessor.getPropertyValue(AbstractNestablePropertyAccessor.java:617) ~[spring-beans-5.3.7.jar:5.3.7]
	at org.springframework.validation.AbstractPropertyBindingResult.getActualFieldValue(AbstractPropertyBindingResult.java:104) ~[spring-context-5.3.7.jar:5.3.7]
	at org.springframework.validation.AbstractBindingResult.getFieldValue(AbstractBindingResult.java:228) ~[spring-context-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.support.BindStatus.<init>(BindStatus.java:129) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.getBindStatus(AbstractDataBoundFormElementTag.java:178) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.getPropertyPath(AbstractDataBoundFormElementTag.java:199) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.getName(AbstractDataBoundFormElementTag.java:164) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.autogenerateId(AbstractDataBoundFormElementTag.java:149) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.resolveId(AbstractDataBoundFormElementTag.java:139) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractDataBoundFormElementTag.writeDefaultAttributes(AbstractDataBoundFormElementTag.java:122) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractHtmlElementTag.writeDefaultAttributes(AbstractHtmlElementTag.java:460) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.InputTag.writeTagContent(InputTag.java:357) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.form.AbstractFormTag.doStartTagInternal(AbstractFormTag.java:87) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.tags.RequestContextAwareTag.doStartTag(RequestContextAwareTag.java:83) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.apache.jsp.WEB_002dINF.views.data_002dman.chargingProfileAdd_jsp._jspx_meth_form_005finput_005f8(chargingProfileAdd_jsp.java:1170) ~[steve.jar:?]
	at org.apache.jsp.WEB_002dINF.views.data_002dman.chargingProfileAdd_jsp._jspx_meth_c_005fforEach_005f1(chargingProfileAdd_jsp.java:1094) ~[steve.jar:?]
	at org.apache.jsp.WEB_002dINF.views.data_002dman.chargingProfileAdd_jsp._jspx_meth_form_005fform_005f0(chargingProfileAdd_jsp.java:414) ~[steve.jar:?]
	at org.apache.jsp.WEB_002dINF.views.data_002dman.chargingProfileAdd_jsp._jspService(chargingProfileAdd_jsp.java:256) ~[steve.jar:?]
	at org.apache.jasper.runtime.HttpJspBase.service(HttpJspBase.java:71) ~[apache-jsp-8.5.54.jar:8.5.54]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder$NotAsync.service(ServletHolder.java:1443) ~[jetty-servlet-9.4.40.v20210413.jar:9.4.40.v20210413]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:791) ~[jetty-servlet-9.4.40.v20210413.jar:9.4.40.v20210413]
	at org.eclipse.jetty.servlet.ServletHandler$ChainEnd.doFilter(ServletHandler.java:1626) ~[jetty-servlet-9.4.40.v20210413.jar:9.4.40.v20210413]
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:548) ~[jetty-servlet-9.4.40.v20210413.jar:9.4.40.v20210413]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) ~[jetty-server-9.4.40.v20210413.jar:9.4.40.v20210413]
[... 224 lines cut ...]
	at org.springframework.web.servlet.view.AbstractView.render(AbstractView.java:316) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.DispatcherServlet.render(DispatcherServlet.java:1396) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.DispatcherServlet.processDispatchResult(DispatcherServlet.java:1141) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1080) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:963) ~[spring-webmvc-5.3.7.jar:5.3.7]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.7.jar:5.3.7]
	... 78 more
```

</details>

Yes, my database was correctly modified on update:
```
MariaDB [(none)]> use stevedb; describe charging_schedule_period;
Database changed
+-------------------------+---------------+------+-----+---------+-------+
| Field                   | Type          | Null | Key | Default | Extra |
+-------------------------+---------------+------+-----+---------+-------+
| charging_profile_pk     | int(11)       | NO   | PRI | NULL    |       |
| start_period_in_seconds | int(11)       | NO   | PRI | NULL    |       |
| power_limit             | decimal(15,1) | NO   |     | NULL    |       |
| number_phases           | int(11)       | YES  |     | NULL    |       |
+-------------------------+---------------+------+-----+---------+-------+
4 rows in set (0.001 sec)
```

goekay, 2021-05-20:
cannot reproduce. it's working fine with
- win 7
- jdk-11.0.11+9 from AdoptOpenJDK
- mariadb-10.4.18

the usage of `powerLimitInAmperes` in your stacktrace suggests that your git tree does not include changes made in https://github.com/RWTH-i5-IDSG/steve/pull/553 to fix the referenced issue there.

i think your git tree does not include some upstream changes.

barsnick, 2021-05-25:
Indeed, something is very fishy. Even though ```git status``` and ```git diff``` show me that everything is fine (I'm on 252337af01128d5e805fbff798be6deed76a1511 which is identical with steve-3.4.5), the git build behaves differently than your newly released 3.4.5, built from tar.gz.

Sorry for the noise. I'm closing this.

closed 2021-05-25 by barsnick