steve-community/steve #585: Question [Update JSP File, but not change]

caktozz, 2021-05-21:
Hi, I am just learning and want to implement this technology with my project on communication protocols for charging on electric vehicles.

I've created a Steve Server using AWS Lightsail and it's running well. however, I want to make changes to the web interface.

I did remote using WinSCP and made changes to the JSP file. Changes were successful on WinSCP and Steve Servers. however, the web interface has not changed. How to solve this? I've tried restarting the Steve Server but it doesn't change. Thank's

goekay, 2021-05-21:
you need to rebuild steve.

closed 2021-05-25 by goekay

caktozz, 2021-05-25:
I try to rebuild with "mvn package", and result is "Error"
"Failed to execute goal org.eclipse.jetty:jetty-jspc-maven-plugin:9.4.40.v20210413:
jspc (jspc)"

I still try and i don't know how to solve this

caktozz, 2021-05-25:
Ok, Thank you for your help