steve-community/steve #586: Question - OCPP Tag / User management understanding

RedPeace, 2021-05-21:
Hi,

i have a SteVe Server up and running.
Actually two wallboxes connected. In the future there should be four boxes.

But i don't understand how i can assign a user to a box so that the user get's e-mail notifications 
or can logon to the SteVe server to see some informations about his box.

I've only one OCPP Tag named "zero". This was imho auto generated after first box connected to SteVe.
And if i create a new user i can't choose a box but only a OCPP tag.
If i try to manually create a OCPP tag i can't choose a existing wallbox.

So i'm a litte bit confused how to setup users :/

Maybe anyone can help me please? :/

Regards.
Gero

goekay, 2021-05-21:
funcionality-wise, steve and ocpp only care for the charging stations and ocpp tags. we introduced _the user model_ later to associate an ocpp tag with more information. nothing more than that. it's a meta thing. therefore, you can even have a user with multiple ocpp tags.

a user cannot be assigned to a station/box. a user within the steve world can use any box.

email notifications are not meant for the end users. they are meant for the maintainer of stations (the user of steve if you will) . therefore, the notifications will be sent for any interaction with all stations. it's not what you expect them to be.

> ... or can logon to the SteVe server to see some informations about his box.

you hopefully followed the configuration steps and changed [the web interface credentials](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L17-L18). you can logon using these.

RedPeace, 2021-05-21:
Thank you for clarification!

So i can't use SteVe as expected and have to reconsider :)

Can you explain shortly what a OCPP Tag is and what can be done with it?
Or is there any manual for SteVe? :)

Yes, i've changed the credentials. But thank you for the hint ;)

goekay, 2021-05-21:
> Can you explain shortly what a OCPP Tag is and what can be done with it?

ocpp tag refers to an rfid tag/card. it can be used for authorization and identification of charging sessions. please take a look at the ocpp specification for details.

> Or is there any manual for SteVe? :)

no there is not.

RedPeace, 2021-05-21:
OK, thank you again!

closed 2021-05-21 by RedPeace