steve-community/steve #596: API REST MeterValuesSampledData

sea-blorin, 2021-05-28:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
How can I acquire the "MeterValuesSampledData" data, for example: SoC, Power.Active.Export, Power.Active.Import, Energy.Active.Export.Register, Energy.Active.Import.Register, Frequency, by API REST in JSON format from STEVE?
...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2021-05-28:
this is currently not possible, because it is not implemented. however, you can implement it easily. this is not a big task, if you know your way around java, databases and rest apis in spring.

or, you can always go to the database directly and produce an export.

sea-blorin, 2021-05-31:
Perfect, but what is the method to extract the Meter Values Data?

goekay, 2021-05-31:
i am not sure whether you are asking this in more abstract terms, or something like "the java method/code that does the work".

sea-blorin, 2021-06-09:
@ksushant881 How you get a single meter value from CP? Can you explain me?

sea-blorin, 2021-06-14:
What is the method to receive SoC, Power.Active.Export, Power.Active.Import, Energy.Active.Export.Register, Energy.Active.Import.Register, Frequency from SteVe?

sea-blorin, 2021-06-14:
I've found this: https://github.com/RWTH-i5-IDSG/steve/blob/72e086e496c46babe13c8c0c01f0126010f7108f/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java

goekay, 2021-06-14:
the part after "Step 2: Collect intermediate meter values" is the relevant code.

sea-blorin, 2021-06-15:
@goekay is it a way to active this in the web interface?

goekay, 2021-06-15:
it is active. 

you can click on a transaction id to go to the "transaction details" page. on this page you can see the intermediate meter values on a table.

sea-blorin, 2021-06-15:
@goekay thanks, but on this page I can't see nothing in data table

sea-blorin, 2021-06-15:
[image]

goekay, 2021-06-15:
then... it means that the charging station does not report any meter values. steve shows everything that it receives in this table. we do not do any filtering.

sea-blorin, 2021-06-17:
Ok! But for the firmware update, "Location URI" what does it indicate? If for example I've the firmware file on a FTP server?
[image]

mzsolnay, 2025-10-07:
I am not sure if i understood the original task described at the beginning, but I guess I had something similar in mind and came to following solution. I run a Node-Red instance next to Steve. I check the SteVe logs with an exec-node, and progress the incoming new lines. I can extract information like actual charging power. There is a "MeterValueSampleInterval" parameter in the chargers configuration, that i set to 30, so during a charge session i get actual chargng power (for each phase) every 30 sec. I hope I could help !