steve-community/steve #599: Webapp stopped - bug or feature? -HikariPool Shutdown | node0 Stopped scavenging

christophb1984, 2021-06-08:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.5
Operating system  : Ubuntu 20.4
JDK               : (build 11.0.11+9-Ubuntu-0ubuntu2.20.04)
Database          : 10.3.29-MariaDB-0ubuntu0.20.04.1
```

#### Expected Behavior
The webapp should run 24/7

#### Actual Behavior
Webapp shuts down if there are no CP connections or no active access to the webapp. 

#### Steps to Reproduce the Problem

  1. Wait some time (15 to 20 min. after last active connection)

#### Additional context (LOG)
[WARN ] 2021-06-07 15:15:49,866 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=HELLOHELLO, sessionId=c85476c6-da21-ce83-cb49-eced921f8898] Connection is closed, status: CloseStatus[code=3001, reason=null]
[INFO ] 2021-06-07 15:16:01,349 org.eclipse.jetty.server.AbstractConnector - Stopped ServerConnector@2e9f0454{HTTP/1.1, (http/1.1)}{192.168.180.10:8080}
[INFO ] 2021-06-07 15:16:01,350 org.eclipse.jetty.server.session - node0 Stopped scavenging
[INFO ] 2021-06-07 15:16:01,354 org.eclipse.jetty.server.handler.ContextHandler.steve - Destroying Spring FrameworkServlet 'spring-dispatcher'
[INFO ] 2021-06-07 15:16:01,365 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown initiated...
[INFO ] 2021-06-07 15:16:01,373 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown completed.
[INFO ] 2021-06-07 15:16:01,375 org.eclipse.jetty.server.handler.ContextHandler.steve - Closing Spring root WebApplicationContext
[INFO ] 2021-06-07 15:16:01,376 org.eclipse.jetty.server.handler.ContextHandler - Stopped o.e.j.w.WebAppContext@3df496d9{/steve,jar:file:/home/christoph/steve-steve-3.4.5/target/steve.jar!/webapp,STOPPED}

goekay, 2021-06-08:
> Webapp shuts down if there are no CP connections or no active access to the webapp.

are you sure about this?

from the log, it looks like that the server is stopping for whatever reason. _as a result_, spring context is shutting down triggering shut down of hikari pool as well.

i locally started steve and killed mysql afterwards. steve survived and continued to live with the following errors in log:

```
[WARN ] 2021-06-08 11:55:02,169 com.zaxxer.hikari.pool.PoolBase (qtp460741164-18) - HikariPool-1 - Failed to validate connection com.mysql.cj.jdbc.ConnectionImpl@23eff4a1 (No operations allowed after connection closed.). Possibly consider using a shorter maxLifetime value.
[DEBUG] 2021-06-08 11:55:32,151 org.jooq.tools.LoggerListener (qtp460741164-18) - Exception                
org.jooq.exception.DataAccessException: Error getting connection from data source HikariDataSource (HikariPool-1)
	at org.jooq_3.14.11.MYSQL.debug(Unknown Source) ~[?:?]
	at ...
```

you might want to look into why the server is being stopped.

christophb1984, 2021-06-08:
Thanks for quick response. Do you have any idea why the service might be stopped? 
The server itself is running. E.g. reaching via SSH is no problem. 
I did no more than install/setup everything as the[ install guide](https://github.com/RWTH-i5-IDSG/steve/wiki/Prepare-Ubuntu-VM-for-SteVe) shows. The Ubuntu server's running on ESXI VMWARE. Might this be a problem?

goekay, 2021-06-08:
hard to say, really. 

do you observe a lot of disconnects/reconnects with stations? if it is the case, maybe a memory leak causes allocating more and more memory until the limit is reached, upon which the system is killing the application. related issue: https://github.com/RWTH-i5-IDSG/steve/issues/509

can you monitor the memory usage of steve through time?

christophb1984, 2021-06-08:
Thanks for he hint. I'm going to check.

christophb1984, 2021-06-09:
> 

> 

    Webapp shuts down if there are no CP connections or no active access to the webapp.

> are you sure about this?

Guess it was a Linux beginner mistake. The SSH session expired and the SteVe.jar Service shutdown after the SSH session shuts down and the webservice died. 
Solution seems to be: [nohub](https://manpages.ubuntu.com/manpages/focal/man1/nohup.1.html) or [tmux](https://manpages.ubuntu.com/manpages/focal/en/man1/tmux.1.html) 

Maybe this hint could be part of the Ubuntu guide?!

closed 2021-06-09 by christophb1984

goekay, 2021-06-09:
>Maybe this hint could be part of the Ubuntu guide?!

FYI: @csamsel