steve-community/steve #605: Default credentials?

glynhudson, 2021-06-17:
I'm successfully running steve via the docker container and can access the UI on port 8180. However, there is no option to create a new account, only a login option. What is the default username/password? Sorry, I know this is a silly question, but I can't see it mentioned anywhere in the docs.

Thanks, keep up the great work 👍

glynhudson, 2021-06-17:
Found them: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/docker/main.properties#L17

closed 2021-06-17 by glynhudson

goekay, 2021-06-17:
>  I know this is a silly question, but I can't see it mentioned anywhere in the docs.

3rd bullet of step 3 mentiones the credentials: https://github.com/RWTH-i5-IDSG/steve#configuration-and-installation

glynhudson, 2021-06-17:
The default username is `admin` and default password `1234`, as far as I can see these default credentials are not mentioned in the readme.