steve-community/steve #607: getting maven build error

Maha-Dev95, 2021-06-17:
#### Checklist

> Getting maven build error in ubuntu even after changes made in the main.properties as given in the steve guidlines:
For more detail please have look in below details,

ubuntu@ubuntu1804:~/steve-steve-3.4.1$ mvn package
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[INFO] Scanning for projects...
[INFO] 
[INFO] -------------------------< de.rwth.idsg:steve >-------------------------
[INFO] Building steve 3.4.1
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-java) @ steve ---
[INFO] 
[INFO] --- git-commit-id-plugin:3.0.1:revision (default) @ steve ---
[INFO] 
[INFO] --- properties-maven-plugin:1.0.0:read-project-properties (default) @ steve ---
[INFO] 
[INFO] --- flyway-maven-plugin:5.2.4:migrate (default) @ steve ---
[INFO] Flyway Community Edition 5.2.4 by Boxfuse
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.549 s
[INFO] Finished at: 2021-06-17T08:55:31-04:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:5.2.4:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: 
[ERROR] Unable to obtain connection from database (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 08S01
[ERROR] Error Code : 0
[ERROR] Message    : Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] : Connection refused (Connection refused)
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

goekay, 2021-06-17:
the exception is very clear:

> [ERROR] Unable to obtain connection from database (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
> ...
> [ERROR] SQL State : 08S01
> [ERROR] Error Code : 0
> [ERROR] Message : Communications link failure
> [ERROR]
> [ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
> [ERROR] : Connection refused (Connection refused)

steve cannot access your db and obtain a connection.

Maha-Dev95, 2021-06-18:
Thanks for your speedy revert.

Can you please explain me, how to connect steve to database and what are
changes need to be made?

Thanks & Regards
mahadev N

csamsel, 2021-06-18:
Just follow the instructions

goekay, 2021-06-25:
closing this issue because there was no follow-up question/discussion.

closed 2021-06-25 by goekay