steve-community/steve #608: Java error when adding ChargingProfile

andle, 2021-06-17:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.5
Operating system  : Raspian
JDK               : 11.1
Database          : 1.0.3
```

Just pulled todays code and built. I am trying to add a ChargePointMaxProfile as seen below, and I am getting a Java error. This worked a few weeks ago.
[image]

#### Expected Behavior
Trying to add MaxChargingProfile, expect profile to be sent to EVSE
...

#### Actual Behavior
Java error
...

#### Steps to Reproduce the Problem

  1. try to add Maxprofile, any values, filling out valid fields, at least one schedule period
  2. click Add
  3. see java error

#### Additional context
<h2 style="color: rgb(0, 0, 0); font-family: &quot;Times New Roman&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">HTTP ERROR 500 org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?</h2>
URI: | /steve/manager/chargingProfiles/add
-- | --
500
org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?
spring-dispatcher
org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?
org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?

<h3 style="color: rgb(0, 0, 0); font-family: &quot;Times New Roman&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Caused by:</h3><pre style="color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.beans.NotReadablePropertyException: Invalid property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' of bean class [de.rwth.idsg.steve.web.dto.ChargingProfileForm]: Bean property 'schedulePeriodMap[zz1623963549536].powerLimitInAmperes' is not readable or has an invalid getter method: Does the return type of the getter match the parameter type of the setter?
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1014)
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:909)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:799)
	at org.eclipse.jetty.servlet.ServletHandler$ChainEnd.doFilter(ServletHandler.java:1626)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:327)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:115)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:81)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:121)
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:115)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:126)
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:81)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:105)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:149)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:63)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.authentication.www.BasicAuthenticationFilter.doFilterInternal(BasicAuthenticationFilter.java:149)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:119)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:336)
	at org.springframework.security.web.authentication.AbstractAuthenticationProcessingFilter.doFilter(AbstractAuthenticationProcessingFilter.java:218)
[... 87 lines cut ...]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:620)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:235)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1624)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:233)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1435)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:188)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:501)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1594)
[... cut ...]
	at org.springframework.web.servlet.DispatcherServlet.render(DispatcherServlet.java:1396)
	at org.springframework.web.servlet.DispatcherServlet.processDispatchResult(DispatcherServlet.java:1141)
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1080)
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:963)
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006)
	... 87 more</pre>
...

goekay, 2021-06-17:
pls see https://github.com/RWTH-i5-IDSG/steve/issues/581

goekay, 2021-06-25:
closing this issue because there was no follow-up question/discussion and it was a duplicate.

closed 2021-06-25 by goekay