steve-community/steve #615: Meter reading values without transaction

untouch68, 2021-07-05:
Hey,

first of all, awesome project! It helps me a lot for development!

My problem that I have is, when the EVcharger is connected and I start charging SteVe only shows me the connector status as "available". If I use the option in the OCPP commands with Start Transaction, it shows me that its charging and tells me meter readings etc in the transaction.

But how do I get the current meter reading of the EVcharger even if no transaction is recognized by SteVe.

For example, if I setup the box that it starts charging by just plugging in the cable, SteVe does not recognize any changes and does not show me information except "available".

I know that this information is stored somewhere on the box, because if I manually start, it shows me the updated meter reading in the transaction, I just don't know how to access it via OCPP.

I am sorry for my english and hope this was kind of understandable.

Greetings!

goekay, 2021-07-06:
in general: most of the time steve relies and reacts to messages coming from a charger. if the message does not come, steve cannot do much about it.... mostly. if you have unexpected behaviour, please observe the incoming messages from the charger. that being said:

> My problem that I have is, when the EVcharger is connected and I start charging SteVe only shows me the connector status as "available".
 
by default, steve expects a ConnectorStatus message after a StartTransaction message. the ConnectorStatus message should and will report this connector as busy. afterwards, the connector will not be available in steve anymore. however, we have a configuration flag to change this logic. for each station you can set a yes/no configuration for "Insert connector status after start/stop transaction". if this is enabled, steve itself will set it as busy without waiting on a ConnectorStatus from charger.

> But how do I get the current meter reading of the EVcharger even if no transaction is recognized by SteVe.

> For example, if I setup the box that it starts charging by just plugging in the cable, SteVe does not recognize any changes and does not show me information except "available".

it's not that a _transaction is not recognized by steve_. probably the charger does not send a StartTransaction. in my message above i was commenting on "ConnectorStatus message not arriving". your problem seems to be that a StartTransaction is not being sent from the charger.

goekay, 2021-07-25:
i consider this question as answered since there were no replies, and therefore am closing the issue.

closed 2021-07-25 by goekay