steve-community/steve #632: Detected failed migration to version 0.6.6 (inital) - How to fix this?

LordDarkneo, 2021-07-30:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.5
Operating system  : Docker
JDK               : AdoptOpenJDK11
Database          : MariaDB
```

#### Expected Behavior

Application built through docker

#### Actual Behavior

After fixin the stroe procedure exection (by connecting to the DB and changing the value `SET GLOBAL log_bin_trust_function_creators = 1;`) I am no having the error message:

```
app_1  | [INFO] --- flyway-maven-plugin:7.9.1:migrate (default) @ steve ---
app_1  | [INFO] Flyway Community Edition 7.9.1 by Redgate
app_1  | [INFO] Database: jdbc:mysql://mariadb:3306/stevedb (MySQL 5.5)
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] BUILD FAILURE
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] Total time:  20.811 s
app_1  | [INFO] Finished at: 2021-07-30T09:29:09Z
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.9.1:migrate (default) on project steve: org.flywaydb.core.api.exception.FlywayValidateException: Validate failed: Migrations have failed validation
app_1  | [ERROR] Detected failed migration to version 0.6.6 (inital). Please remove any half-completed changes then run repair to fix the schema history.
app_1  | [ERROR] Need more flexibility with validation rules? Learn more: https://flywaydb.org/custom-validate-rules

```
 
#### Steps to Reproduce the Problem

  1. Try a first build with latest MariaDB version
  2. You will have the error `[ERROR] Message : You do not have the SUPER privilege and binary logging is enabled (you *might* want to use the less safe log_bin_trust_function_creators variable)`
  3. Connect to the database container in SSH, and run: `mysql -u root -p` enter your root password, and then execute the SQL command: `SET GLOBAL log_bin_trust_function_creators = 1;`
  4. Quit mysql/ssh and try to rebuild the SteVe application
  5. You will have the error message : 
```
app_1  | [INFO] --- flyway-maven-plugin:7.9.1:migrate (default) @ steve ---
app_1  | [INFO] Flyway Community Edition 7.9.1 by Redgate
app_1  | [INFO] Database: jdbc:mysql://mariadb:3306/stevedb (MySQL 5.5)
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] BUILD FAILURE
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] Total time:  20.811 s
app_1  | [INFO] Finished at: 2021-07-30T09:29:09Z
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.9.1:migrate (default) on project steve: org.flywaydb.core.api.exception.FlywayValidateException: Validate failed: Migrations have failed validation
app_1  | [ERROR] Detected failed migration to version 0.6.6 (inital). Please remove any half-completed changes then run repair to fix the schema history.
app_1  | [ERROR] Need more flexibility with validation rules? Learn more: https://flywaydb.org/custom-validate-rules
app_1  | [ERROR] -> [Help 1]
app_1  | [ERROR]
app_1  | [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
app_1  | [ERROR] Re-run Maven using the -X switch to enable full debug logging.
app_1  | [ERROR]
app_1  | [ERROR] For more information about the errors and possible solutions, please read the following articles:
app_1  | [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
steve-345_app_1 exited with code 1

```

#### Additional context

Docker on RPI

goekay, 2021-08-02:
i could not reproduce the problem on a macbbook pro using
- adoptopenjdk11 and
- mariadb 10.6.3,
- with the latest steve (no dirty head) & no existing databases/users,
- after executing the steps from our readme first.

LordDarkneo, 2021-08-03:
Hello and thanks for your feedback,
Are you using containers to reproduce the issue? Because my major issue is that I cannot flyway repair the container, as the build is failing....

goekay, 2022-01-23:
hey, sorry for overlooking this. i am not sure whether you solved your issue already, but...

you can modify the Dockerfile with additional commands to execute additional steps. for ex, you can convert [this line](https://github.com/RWTH-i5-IDSG/steve/blob/master/Dockerfile#L27) into

```
mvn clean flyway:repair package -Pdocker -Djdk.tls.client.protocols="TLSv1,TLSv1.1,TLSv1.2" && \
```

to repair the database, or

```
mvn clean flyway:clean package -Pdocker -Djdk.tls.client.protocols="TLSv1,TLSv1.1,TLSv1.2" && \
```

to drop everything in db for a fresh start. use this clean command with care.

closed 2022-01-23 by goekay

naerschhersch, 2023-03-05:
should be the same issue here

your modification shown above does not have any effect on my side.

```
app_1  | [INFO] --- flyway-maven-plugin:7.15.0:migrate (default) @ steve ---
app_1  | [INFO] Flyway Community Edition 7.15.0 by Redgate
app_1  | [INFO] Database: jdbc:mysql://mariadb:3306/stevedb (MySQL 5.5)
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] BUILD FAILURE
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [INFO] Total time:  11.744 s
app_1  | [INFO] Finished at: 2023-03-05T13:51:02Z
app_1  | [INFO] ------------------------------------------------------------------------
app_1  | [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.api.exception.FlywayValidateException: Validate failed: Migrations have failed validation
app_1  | [ERROR] Detected failed migration to version 0.6.6 (inital). Please remove any half-completed changes then run repair to fix the schema history.
app_1  | [ERROR] Need more flexibility with validation rules? Learn more: https://rd.gt/3AbJUZE
app_1  | [ERROR] -> [Help 1]
app_1  | [ERROR]
app_1  | [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
app_1  | [ERROR] Re-run Maven using the -X switch to enable full debug logging.
app_1  | [ERROR]
app_1  | [ERROR] For more information about the errors and possible solutions, please read the following articles:
app_1  | [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
steve-steve-350_app_1 exited with code 1
```