steve-community/steve #635: Übersetzung einzelner Beschreibungen

LadesaeulenBetreiber96450, 2021-07-31:
Hallo an alle,

Ist es möglich die HTML-Oberfläche des Steve Servers in eine andere Sprache zu übersetzten.
Wäre dankbar wenn es Lösungsvorschläge geben würde.

Gruß Dominik

goekay, 2021-08-02:
yes, it is. you can google for "internationalization spring" to see the possibilities and guides. this would mean converting every bit of english text we have right now into a variable whose actual value (depending on locale/language) needs to be stored in a distinct messages file.

i see it as a big undertaking and am not sure about its cost-to-benefit ratio.

closed 2021-08-02 by goekay

csamsel, 2021-08-03:
a different approach would be to just translate the jsp pages inside the WEB-INF directory for their personal use.