steve-community/steve #636: Unable to obtain connection from database

aatronics, 2021-08-01:
When I run the below command I get "build failed message", anyone
[image]
[image]
 can please help me to run the service

export MAVEN_OPTS="-Xmx100m" && mvn package

csamsel, 2021-08-01:
The error indicates that the password or username is not correct.

aatronics, 2021-08-01:
> The error indicates that the password or username for the database is not correct.

I have used the same username and password to create the database, also I have crosscheck the username and password on Connection details (AWS database).
is this issue will be related to timezone, because whenever I will try to open SSH from the browser, I am not able to connect to Ubuntu VPS, so always I use putty SSH to connect

cyrilfr, 2021-08-23:
I have the same issue using Docker. In my case, the IP used to connect to the DB is the SteVe's Docker IP instead of the `db.ip` value set in the docker configuration file.

joseapdmota, 2023-01-27:
Hi Guys,
Just run into this issue.

How did you overcome?

I'm using the docker method.

Thanks

LivingWithHippos, 2023-09-06:
My error was editing the `main.properties` file under the `prod` folder instead of `docker`

closed 2024-02-18 by goekay