steve-community/steve #641: Documentation

MysticJay, 2021-08-10:
The below checklist asks if I did read the documentation...
Actually I fail to find the documentation, unless it is "just" the Wiki in here?!

1) So could you add a link to some kind of DOC?
2) is there a forum or better a chat (Telegram?) to sort out some (basic) questions?

BR
M.
 

#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2021-08-10:
> ..., unless it is "just" the Wiki in here?!

yes, it is. the [readme](https://github.com/RWTH-i5-IDSG/steve/blob/master/README.md) and the wiki pages hopefully cover some basics.

since steve is an implementation of ocpp, we assume that the reader is familiar with official ocpp spec/documentation regarding the protocol, flows and sequences as well.

> 1. So could you add a link to some kind of DOC?
> 2. is there a forum or better a chat (Telegram?) to sort out some (basic) questions?

there is none.

what is your (basic) question?!

MysticJay, 2021-08-11:
While I already send a private mail to you a couple of days ago, my greatest concern:

What is the goal of SteVe? Just to provide a PoC or a "Public Domain" Software?
e.g. I see no User management, no way to change the admin's password.
Is it ready to handle a charging network?
What about accounting?
Is SteVe only meant to collect and store the data?

My setup will start with one Keba X and will be expanded to 16-24 chargepoints.
We will not have roaming. 

So is SteVe ready to use? what needs to be added by me e.g. in Office?

M.

goekay, 2021-08-11:
i will address only one of the arguments to make a point. i hope you can understand it and draw consequences.

> e.g. I see no User management, no way to change the admin's password.

there _is_ a user management. the term _id tags_ (i.e. the tags of the rfid cards) is used in the world of ocpp to authorise and track charging sessions. we added a user model on top of the model of ocpp id tags: users can be associated with ocpp id tags. one of the screenshots at the bottom of the main page showcases user management. did you even look at it before making the claim? let me make it easier for you: [here it is](https://github.com/RWTH-i5-IDSG/steve/blob/master/website/screenshots/users.png)

there _is_ a way to change the admin password, and it is here: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L17-L18

this configuration is referenced on the [readme page under point 3 as "You must change web interface credentials"](https://github.com/RWTH-i5-IDSG/steve#configuration-and-installation). this is the same readme that you belittled when you criticised the project as to not having a documentation. i wonder whether a comprehensive documentation would have helped you if you disregard already existing information.

as with most open source projects the following holds:  there is no company behind steve. there is no professional customer support to deal with and answer basic questions. you seem to require a safety net in order to use steve. we unfortunately do not provide it. take it or leave it.

MysticJay, 2021-08-12:
Sure I read readme, config file and "Point 3", but that is not what I expect.

the "users" linked to ID-Tags are customers, but these should not have access to SteVe!?

With "User Management" I did not mean the CP-Users, but users of SteVe.
I expect not only an Admin-Account, but users that have access to certain areas.
e.g. Roles like configure , accounting, CP-Management (reservations, start/stop of a session)
These roles should be assigned to different accounts.
I guess you now know what I mean.

M.

MysticJay, 2021-09-05:
Still not clear, what I am asking about?