steve-community/steve #644: Implementation with the server

belljohan3, 2021-08-15:
I built the app stev.jar but i can't start it. Can someone help me out ? we can chat on Telegram. Here is my ID: @johanbell. May be a problem with a server ?
[steve.log](https://github.com/RWTH-i5-IDSG/steve/files/6987826/steve.log)

goekay, 2021-08-15:
steve uses the time zone of utc. your mysql instance should use utc as well. this is not the case and the errors in the log file indicate this problem.

belljohan3, 2021-08-15:
Ok ! @goekay. i'm using mariaDB. How can i set it please ? Where can i find the MySQL instance ?

goekay, 2021-08-15:
i added the info to our FAQ since this issue comes up a lot: https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-to-fix-the-exception-mysql-and-java-are-not-using-the-same-time-zone

if you are not familiar with these concepts, i would suggest googling.

belljohan3, 2021-08-15:
OKay !
THanks

belljohan3, 2021-08-15:
Is it here that we have to change the time zone?

[image]

goekay, 2021-08-15:
please do not change the time zone of steve itself, but the time zone of mysql/mariadb. in my message above and in the faq entry i hinted at that. why are you still trying to change the time zone of steve?

changing the time zone of steve will cause issues, since many charging stations cannot handle time zones properly.

belljohan3, 2021-08-15:
Am i on the good way my friend ?

[image]

goekay, 2021-08-29:
i consider the issue resolved, and am closing it.

closed 2021-08-29 by goekay