steve-community/steve #652: Unable to receive a heartbeat from Delta Charger (EVDU25xx)

204maker, 2021-08-30:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.0.2
Operating system  : Windows 10 Education, 21H1
JDK               : 
Database          : 0.9.3
```

#### Expected Behavior

New charge point is able to send heartbeat to SteVe server

#### Actual Behavior

Expectation failed.

#### Steps to Reproduce the Problem

  1. Server IP is default at 192.168.1.69
  2. Delta Charger was configured to use:
      -  OCPP1.6
      - IP address 192.168.1.10 (Gateway: 192.168.1.1, Mask: 255.255.255.0)
  3. Laptop was connected to the network, IP address: 192.168.1.7

#### Additional context
**Latest log file:**

_[ERROR] 2021-08-30 10:46:05,046 de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor - Exception happened
com.ctc.wstx.exc.WstxEOFException: Unexpected EOF in prolog
 at [row,col {unknown-source}]: [1,0]
	at com.ctc.wstx.sr.StreamScanner.throwUnexpectedEOF(StreamScanner.java:687) ~[steve-3.0.2.jar:?]
	at com.ctc.wstx.sr.BasicStreamReader.handleEOF(BasicStreamReader.java:2220) ~[steve-3.0.2.jar:?]
	at com.ctc.wstx.sr.BasicStreamReader.nextFromProlog(BasicStreamReader.java:2126) ~[steve-3.0.2.jar:?]
	at com.ctc.wstx.sr.BasicStreamReader.next(BasicStreamReader.java:1181) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.staxutils.DepthXMLStreamReader.next(DepthXMLStreamReader.java:220) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.staxutils.DepthXMLStreamReader.nextTag(DepthXMLStreamReader.java:232) ~[steve-3.0.2.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:64) ~[steve-3.0.2.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:36) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.phase.PhaseInterceptorChain.doIntercept(PhaseInterceptorChain.java:308) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.ChainInitiationObserver.onMessage(ChainInitiationObserver.java:121) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.http.AbstractHTTPDestination.invoke(AbstractHTTPDestination.java:267) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invokeDestination(ServletController.java:234) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:208) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:160) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.CXFNonSpringServlet.invoke(CXFNonSpringServlet.java:216) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.handleRequest(AbstractHTTPServlet.java:301) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.doGet(AbstractHTTPServlet.java:225) ~[steve-3.0.2.jar:?]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687) ~[steve-3.0.2.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.service(AbstractHTTPServlet.java:276) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:864) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:535) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:146) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:257) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1595) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:255) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1253) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:203) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:473) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1564) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:201) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1155) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.handle(GzipHandler.java:716) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:61) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.Server.handle(Server.java:531) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:352) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:260) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:281) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:102) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:118) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:319) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:175) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:133) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:754) ~[steve-3.0.2.jar:?]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:672) ~[steve-3.0.2.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_181]_

goekay, 2021-08-30:
the station is sending a message that steve cannot handle. the message prematurely ends or something like that. the exception hints at that.

csamsel, 2021-09-01:
Did you maybe mix up the endpoints for JSON/Websockets (OCPP-J) and SOAP (OCPP-S)?

204maker, 2021-09-01:
> Did you maybe mix up the endpoints for JSON/Websockets (OCPP-J) and SOAP (OCPP-S)?

It looks like I mixed up the endpoints. 
I got a heartbeat using OCPP-1.5S and SOAP.

closed 2021-09-04 by goekay