steve-community/steve #653: Architecture not plug and play

techcharzer, 2021-08-31:
#### Checklist

#### Describe the problem you are trying to solve
I cannot plug in and play as it is tightly coupled to the frontend. As I wanted to plug it into a micro-service architecture.

#### Describe the solution you'd like
we should decouple the frontend from the backend and create it as micro-service. If you want to use the fronted just configure the URL and it is ready to be used. But if you want to use only backend then that should be doable. 

...

Mohit7497, 2022-07-19:
Hello, @techcharzer is it doable now to just use the backend of steve?

goekay, 2022-07-19:
hey @Mohit7497 no it is not. front- and backend of steve are still one item.

Mohit7497, 2022-07-19:
hey @goekay do you plan to do it anytime soon or is there a way i could only use the backend. Thanks for your reply.

goekay, 2022-07-19:
> do you plan to do it anytime soon

i do not plan it in the near future. i don't know what the long term can bring.

> is there a way i could only use the backend

you can look at the contoller package and create  REST versions of controllers without tying them to the representation layer (the jsp views).

kohlerdominik, 2022-08-26:
Hi @goekay 

Using Steve as part of a micro service architecture would also be great for us, rather than interfacing with the database directly.

I understand, that this seems no priority of yours. Would you consider to add this as sponsored feature, though? Maybe other users like @techcharzer, @Mohit7497 and more would be open for sponsoring, too.

goekay, 2022-08-30:
> Using Steve as part of a micro service architecture would also be great for us, rather than interfacing with the database directly.

@kohlerdominik can you please reach out to me per mail for us to arrange a meeting. i would like to understand your use cases, requirements and what you expect from this.