steve-community/steve #656: Websocket connection path too long to connect to chargepoint

rishabhmehandru, 2021-09-08:
Hi,

I am testing Steve with Tellus power chargers but the allowed server url length is too short in the device. Is there a way to truncate “ws://ip/steve/websocket/CentralServiceSystem/<device id>“ to make it usable with Tellus Power chargers?

best

goekay, 2021-09-08:
hi,

i assume "websocket/CentralServiceSystem/" was a typo, because the path is not like that.

the postfix is defined here: 
https://github.com/RWTH-i5-IDSG/steve/blob/c2f7ae780e19d91bdaad88dedd79b46a81ccdffe/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L79

you can change `"/websocket/CentralSystemService/"` to something else. but do not forget to keep the asterisk!

rishabhmehandru, 2021-09-08:
Thanks! Thats awesome...I was worried I will have to use some sort of websocket router...this saves work and compute resources

rishabhmehandru, 2021-09-08:
H

Is there somewhere else where I need to modify the path, since build is working but webapp is not able to start? Here is the log error:

 ] 2021-09-08 15:24:24,032 org.eclipse.jetty.server.handler.ContextHandler - Stopped o.e.j.w.WebAppContext@5175d9ad{/steve,jar:file:/home/vagrant/ezstop_steve/steve-steve-3.4.4/target/steve.jar!/webapp,STOPPED}

Best

rishabhmehandru, 2021-09-09:
It got resolved by dropping the database and creating a new one. Tellus chargers were detected. I will add those to the list of compatible chargers after doing some more testing. Thanks! For all the help.

closed 2021-09-09 by rishabhmehandru

iicstoday, 2022-09-27:
Hi please describe which file you change the code I am changing the path and also variable value but still outcomes same.
so please explain.
thanks