steve-community/steve #659: WILL NOT BUILD - SSL Cert expired on dependency

Loop-Donsky, 2021-09-15:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.2
Operating system  : ...
JDK               : JAR 11.0.8
Database          : MariaDB
```

#### Expected Behavior

Will build

#### Actual Behavior

Does not build

#### Steps to Reproduce the Problem

  1. .Attempt a build

#### Additional context
ERROR:
"Failed to execute goal on project steve: Could not resolve dependencies for project de.rwth.idsg:steve:jar:3.4.2: Failed to collect dependencies at de.rwth.idsg:ocpp-jaxb:jar:0.0.1: Failed to read artifact descriptor for de.rwth.idsg:ocpp-jaxb:jar:0.0.1: Could not transfer artifact de.rwth.idsg:ocpp-jaxb:pom:0.0.1 from/to TreibhausNexus (https://treibhaus.informatik.rwth-aachen.de/nexus/repository/maven-public/): PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed: NotAfter: Tue Sep 14 15:44:44 UTC 2021 -> [Help 1]"

Loop-Donsky, 2021-09-15:
Submitted request to root domain and received the following response ( We're in the queue ):
```
Sehr geehrte Damen und Herren,

Ihre E-Mail mit dem Betreff:
            Expired SSL Cert: treibhaus.informatik.rwth-aachen.de
ist bei uns eingegangen und wird in Kürze bearbeitet werden.

Ihre E-Mail hat von uns die Ticket-ID
            DENIC#: 2021091543003521
erhalten. Bitte geben Sie diese bei Rückfragen stets im Betreff mit an.

Ihre DENIC eG

------------------------------------------------------------------

Dear Customer,

your e-Mail with the subject:
            Expired SSL Cert: treibhaus.informatik.rwth-aachen.de
has been received and will be processed within a short time.

Your e-Mail has been given the ticket-ID:
         DENIC#: 2021091543003521
For further queries regarding your e-Mail always mention the ticket-ID in the subject.

DENIC eG

---
Direct Services
```

Loop-Donsky, 2021-09-15:
I redirected to 'http' instead of 'https' this is not a long term solution, but feel free to update the dep URI to not have TLS

goekay, 2021-09-15:
hey, you checked `I am running the latest version and the issue still occurs` even though you are using `SteVe Version     : 3.4.2`.

we saw this coming, and discussed possible solutions in https://github.com/RWTH-i5-IDSG/steve/issues/242

the next release https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-3.4.3 addresses this exact issue.

Loop-Donsky, 2021-09-15:
Ah yes, I thought I was running the latest. Good catch. Thank you.

We're unable to merge downstream right now as we've made some very drastic changes to include a message queue. I will look at the solution in #242, and the related TAG 3.4.3, and try to bring them over.

Appreciate timely the response

closed 2021-09-15 by Loop-Donsky

goekay, 2021-09-15:
version 3.4.3, compared to 3.4.2, just updated the dependencies and switched to jitpack. therefore, i would consider it as an easy upgrade.

Loop-Donsky, 2021-09-15:
I just looked, yeah it seems like it should be smooth. I'm actually going to try and catch up, then discuss the 'proprietary differentiators' with my team, and maybe make a PR back in to support the community we're been participating in.