steve-community/steve #660: 2-Tier Connection Support in Steve

HARSHAL93, 2021-09-16:
Appreciated to all efforts to provide great platform.
I have implemented java application with provided code and everything is working as expected.
As I am new in this domain only work experience in Embedded C with Hardware stuffs.
I am planning to add another one charging connector with same charge point  to make 2-tier charger point,it will be helpful to get support to know from where to start and what kind of changes are required in existing code.

goekay, 2021-09-16:
hey, thanks for the kind words.

if i understand you question correctly, there are no changes necessary. steve supports multiple charging points, and also multiple connectors per charging point.

goekay, 2021-10-03:
i consider this issue resolved.

closed 2021-10-03 by goekay

HARSHAL93, 2021-11-22:
Yes, Thanks